RF1000community/Repetier-Firmware #4: Vorschlag: Single-Step-Höhenänderung in Z eindeutig mit Einheit definieren

Nibbels, 2017-03-10:
Anstatt "ein Step" darzustellen, sollte man 5 Mikrometer, 10Mikrometer oder ein anderer nachvollziehbarer Wert mit Einheit definieren.
Evtl. diese Schrittweite übers Menü umstellbar machen.

mhier, 2017-03-14:
Unbeding! Zuman ein "Step" für mich wie ein Motorschritt klingt, und das stimmt in der Standard-Einstellung überhaupt nicht. Umschaltbar im Menü wäre sehr gut, als kleinste Einstellung sollten unbedingt einzelne Motorschritte möglich sein, das brauche ich oft im Fräs-Betrieb.

Nibbels, 2017-03-18:
Es gibt eine Variable:
g_nManualSteps[Z_AXIS]

`
unsigned long   g_nManualSteps[4]           = { (unsigned long)DEFAULT_MANUAL_STEPS_X, (unsigned long)DEFAULT_MANUAL_STEPS_Y, (unsigned long)DEFAULT_MANUAL_STEPS_Z, (unsigned long)DEFAULT_MANUAL_STEPS_E };`

Die entspricht dem Single_Step der Tasten.
RF.cpp::~12098
```
        case MOVE_MODE_SINGLE_STEPS:
        {
            long    Temp; //bringt nur was wenn homed
            steps = g_nManualSteps[Z_AXIS] * increment;
```

Sie kann nachträglich geändert werden durch:
**M3100 [S] - configure the number of manual z steps after the "Z up" or "Z down" button has been pressed**

Und zwar von 1 bis MAXIMAL_MANUAL_STEPS_Z  
wobei
`#define MAXIMAL_MANUAL_STEPS_Z              (ZAXIS_STEPS_PER_MM *10)`

Der Standardwert ist: 
`#define DEFAULT_MANUAL_STEPS_Z              (RF_MICRO_STEPS *2)`
siehe:
```
/** \brief Define the to-be-used micro steps */
#define RF_MICRO_STEPS                      32    
```

Also ist DEFAULT_MANUAL_STEPS_Z  = 64
demnach stimmt das mit @RF1000 's Aussage überein.

Zusätzlich gilt:
```
/** \brief Number of steps for a 1mm move in z direction  Overridden if EEPROM activated.*/
#define ZAXIS_STEPS_PER_MM                  float(80 * (float)RF_MICRO_STEPS)
```
also 2560 Steps pro Millimeter. Korrekt?
also 0,000390625 mm pro Step
also 0,390625 um pro Step

Will man diese krumme Zahl ca. auf einen merkbaren Mikrometer umrechnen, dann geht z.B. Faktor 5.
Würde man um 5 Steps verfahren hätte man ziemlich genau 2 Mikrometer, aber eben nur fast (1,953125um).

Was fehlt ist das Umstellen per Menü, also 
- ein Menüpunkt
- eine Logik, für die Wertesprünge zwischen 1 und einem Maximum
- eine Merkposition im EEPROM (hab ich bisher nie angefasst!)

LG

Nibbels, 2017-03-19:
Dafür gibts nun ein **flüchtiges** Menü unter 
Position (weiter unten).

Die oben genannten Steps sowie bis 500 um (Mikrometer) sind einstellbar. Stellt man im Menü den Wert um, verhält sich danach "Z-Move-Single-Step" jeweils anders.

EEPROM-Speicherung ist nicht eingebaut.

Siehe:
https://github.com/Nibbels/Repetier-Firmware/commit/d260f2890440679852daad24d9278edc2cca1830

closed 2017-03-19 by Nibbels