RF1000community/Repetier-Firmware #5: Bugfix: Emergency Block durch manuelle Steuerung unmöglich machen

Nibbels, 2017-03-10:
Wenn man mit 1mm, 10mm, 50mm Schritten gehomed in -Z fährt, überfährt man fast mit Sicherheit die Marke Z_OVERRIDE_MAX. Das ist dämlich und sollte verhindert werden. Wirkung: Emergency Block und der Drucker friert (nach Notfall-Plan) ein.

Ebenso:
Wenn man gehomed in -Z fährt, indem man MOVE_MODE_SINGLE_MOVE nutzt, landet man je nach Zufall/Geschwindigkeit in Emergency Block (Because of Z-Min) oder korrekt in "Error Z-Axis Min Reached"

Nibbels, 2017-03-17:
https://github.com/RF1000community/Repetier-Firmware/pull/17

Wäre damit erledigt.

mhier, 2017-03-23:
Im Fräsmodus verhindert das jetzt aber, dass ich definiert über das Menü in negative Z-Bereiche fahren kann. Das ist im Fräsmodus was ganz Normales, denn Z=0 ist die Oberfläche des Objektes und Z<0 liegt im Objekt. Ich bohre z.B. immer mein Referenzloch beim PCB-Fräsen über das Menü, dazu möchte ich auf Z=-2mm fahren.

Sollte man vielleicht eine Ausnahme einbauen, falls der Drucker sich gerade im Fräsmodus befindet?

Nibbels, 2017-03-28:
Ist inzwischen schon als Ausnahme eingebaut, nur aktuell noch ungetestet.

Nibbels, 2017-05-19:
Kann man das schließen? Test erfolgreich?

mhier, 2017-05-22:
Also für mich funktioniert es. Ich schließe es dann mal.

closed 2017-05-22 by mhier