RF1000community/Repetier-Firmware #6: Vorschlag: Diskussion über große Z-Schrittweiten beim Knopfklick

Nibbels, 2017-03-10:
Überlegen, ob größere Schrittweiten (10mm, 50mm) im Printermodus am Knopf nur möglich sein sollen, wenn sich der Nutzer im richtigen Untermenü befindet.
/Position->Position-Z->HIER

Bsp: if (menulevel == a and menupos == b) ...

mhier, 2017-03-14:
Ich würde die beiden Pfeil-Tasten für die Z-Achse von den Menü-Aktionen entkoppeln. Die Pfeil-Tasten sollten immer das gleiche Verhalten haben (single move wenn gerade nicht gedruckt wird, single step wenn gerade gedruckt wird). Nur im Menü ist das Verhalten umschaltbar, was dann auch nur Auswirkungen darauf hat, wenn man über die Menü-Tasten die Z-Achse bewegt.

Das würde dem Verhalten entsprechen, was ich intuitiv erwarten würde.

Das betrifft den Fräs-Modus übrigens genauso.

Nibbels, 2017-03-16:
Merke:

Am PWM_TIMER_VECTOR
hängt UI_FAST (mit &4-Teiler)
hängt uid.fastAction{
- Knöpfe lesen: -> lastButtonAction = KnopfNummer aus ui_check_keys(&action);
- Noch-Immer-Gedrückt lesen: -> g_nContinueButtonPressed = 1
- Knopfe losgelassen lesen: -> PrintLine::stopDirectMove();
}

An Commands.checkforPeriodicalActions()
hängt UI_SLOW
hängt ui.slowAction{
- nextPreviousAction( encoderposition ? ) 
- Check: Menü Autoreturn und noch sowas
- Check: Knopf mehrfach? Knopf-Halten-Timeout? -> executeAction(lastButtonAction)
}

So wirklich interessant scheinen: 
- executeAction(lastButtonAction)
- nextPreviousAction(...)

wobei 
nextPreviousAction(...) ebenfalls executeAction(lastButtonAction) untergeordnet ist, also darin vorkommt,
wenn die Knöpfe "Hoch" = UI_ACTION_PREVIOUS und "Runter" = UI_ACTION_NEXT
als Aktion anstehen.

Wenn wir nach dem Verhalten der Knöpfe ausschau halten, können wir also fast 100% von 
executeAction() bzw. uid.executeAction() als Startpunkt ausgehen.
Ausser es ist eine extrem schnelle Reaktion notwendig, dann von ui.fastAction()

Nibbels, 2017-03-16:
In UIDisplay::**nextPreviousAction**(int8_t next) //next ist die Richtung -1 / 1

- springt man im Menü nach links oder rechts. (Menüebene nach unten oder oben)
- Je nach Entity-Type:  // 0 = Info, 1 = Headline, 2 = submenu ref, 3 = direct action command
hängen auch weitere Direktbefehle drin, die offensichtlich mit Pfeiltaste Rechts aktiviert werden.

Entity-Type 0= Info -> Do -> Return
elseif Entity-Type 1 = Headline -> Do -> Return
elseif Entity-Type 2 = submenu ref -> Do -> Return
elseif Entity-Type 3 = direct action command -> SWITCH -> UI_ACTION_...

Bei der "Richtung" spielt immer dieses INCREMENT mit, welches -1 1 oder 0 zu sein scheint. und per UI_INVERT_INCREMENT_DIRECTION für alle Direktbefehle invertierbar ist.

Nibbels, 2017-03-16:
In UIDisplay::**executeAction**(int action) //action ist der Knopftyp oder eine sonstige "Aktion"

- Wird geprüft ob akteull "Printer::blockAll" aktiv ist. -> Return
- Dann wird geprüft, ob es sich um einen Knopf handelt: -> processButton(action)  -> Return
- Dann werden alle sonstigen Aktionen abgehandelt.  -> Return

Notiz: Diese UI_ACTION_* sind sowohl in **ui.h als auch in RF.h** definiert. Wenn man Glück hat, merkt man das, sodass keine Aktion doppelt definiert wird 🗡 :D 

Beschreibung;
// ----------------------------------------------------------------------------
//                          Action codes
// 1-999     : Autorepeat
// 1000-1999 : Execute
// 2000-2999 : Write code
// 4000-4999 : Show menu
// Add UI_ACTION_TOPMENU to show a menu as top menu
// ----------------------------------------------------------------------------

WICHTIG:

processButton(..) : 

 case UI_ACTION_RF_HEAT_BED_UP:
        {
            nextPreviousZAction( -1 );
            break;
        }
        case UI_ACTION_RF_HEAT_BED_DOWN:
        {
            nextPreviousZAction( 1 );
            break;
        }

nextPreviousAction(..)

        case UI_ACTION_ZPOSITION:
        {
            nextPreviousZAction( increment );
            break;
        }

Sowohl Knopf als auch Menü-Steuerung führen auf dieselbe Funktion:
**nextPreviousZAction**

**-> Dort hatte ich die Mod-Abfrage eingefügt, mit der ich das Z-Offset für den Fall, dass man im Mod-Menü Menü2 ist.
Das ist aber unordendlich, wie mir jetzt scheint.
Ich ändere das und füge den betreffenden Code direkt bei UI_ACTION_RF_HEAT_BED_UP bzw. UI_ACTION_RF_HEAT_BED_DOWN ein.**

Nibbels, 2017-03-17:
Das Thema wäre mit https://github.com/RF1000community/Repetier-Firmware/pull/17 erschlagen, wenn alles auch beim RF1000 funktioniert.

Dort kann ich es leider nicht testen.
Ich kann auch nicht testen, wie sich die Menütasten in den 
Z-Min / Z_Override_Max / Z-Max -Grenzfällen und Homed/Unhomed-Fällen bei den doppelten Endschaltern verhalten.

Für den RF2000 habe ich das pingelig genau druchgemacht.

closed 2017-03-19 by Nibbels