RF1000community/Repetier-Firmware #9: Vorschlag: Homed

Nibbels, 2017-03-10:
Die Bewegung von Z ins Minus hat einiges Bugs:
Generell muss man das nicht 100% ernst nehmen, aber man könnte es. Denn aktuell kann ein unbedarfter Nutzer den Drucker und sich selbst austricksen, ohne wirklich schuld zu sein.

- Homed nur setzen, wenn mindestens X und Y gehomed wurden.
(Gibt es einen Fall, in dem man z.B. für Milling Z nicht homen wollte?)

- Die Z-Kompensation nur aktivieren, wenn mindestens X, Y und Z gehomed wurden.

- Fahren in Z- nur erlauben, wenn mindestens Z gehomed wurde.

- Fahren in Z- ohne homing nur erlauben, wenn der Drucker nicht mit gedrücktem Min-Schalter aufgewacht ist. Der könnte wegen Emergency-Block auf -1mm oder -0.5mm stehen, denken er darf weiter ins Minus und es tun.
Das kollidiert natürlich mit dem Freifahren und der Tatsache, dass der Drucker unten stehen könnte.
Evtl. braucht man im Menü 2 weitere Funktionen mit Sicherheitsabfrage: Nach unten Freifahren, Nach oben Freifahren.

mhier, 2017-03-14:
Ich habe noch nie einzelne Achsen gehomed, von mir aus kann immer alles zusammen gehomed werden :-) Falls es nötig ist, einzelne Achsen zu homen, brauchen wir eigentlich Flags, welche Achsen denn nun gehomed wurden und welche nicht. Alles andere funktioniert nicht sauber. Fahren in Z sollte in beide Richtungen blockiert sein, wenn Z nicht gehomed ist und der Drucker mit gedrücktem Z-Schalter aufwacht. Das ist aber meines Wissens schon der Fall. Freifahren kann man über die Home-Funktion (das knatter zwar, wenn er unten steht, schadet aber nichts). Lediglich das automatische Freifahren macht Probleme, vielleicht sollte man das ganz rausnehmen?

Nibbels, 2017-03-14:
Ich kenn das automatische Freifahren nicht. :D
Dass wir rausgefunden haben, woran es lag, ist rein AtlonXPs Verdienst.
Bei mir im RF2000 sind 2 Schalter auf 2 Pins.
`FEATURE_CONFIGURABLE_Z_ENDSTOPS 0`
deaktiviert das komplette System und stellt es so, wie im RF2000 ein. 

`#define	FEATURE_CONFIGURABLE_Z_ENDSTOPS	1`
gibt 	`DEFAULT_Z_ENDSTOP_TYPE =	[ ENDSTOP_TYPE_SINGLE / ENDSTOP_TYPE_CIRCUIT ]`
erst die Daseinsberechtigung.

Das betrifft demnach rein den RF1000.
Ich könnte versuchen, das Freifahren auszuklinken. Aber das würde ich für den Fall der Fälle dir überlassen. Testen könnte ich es nicht.

Der Suchbegriff wäre: "FEATURE_CONFIGURABLE_Z_ENDSTOPS"

mhier, 2017-03-15:
Sei froh :-) Ich mach das bei Gelegenheit, falls Conrad das nicht bald behebt.

Nibbels, 2017-03-22:
Wie ich das mitbekommen habe, ist heute wieder ein Endschalter gestorben.
Vermutlich wieder Drive-Free.
Vermutlich nicht wegen uns. Eher wegen im Menü Einstellungen ausprobieren und versehentlichem Circuit-Umstellen.

Die Lösung offensichtlich ist meistens: Selbst einen besseren Schalter kaufen und selbst einbauen. Drum bekommt Conrad unter Umständen garnicht so viel da drüber mit?

Nibbels, 2017-07-16:
In der Development 1.37q weiß der Drucker nun genau, welche der Achsen gehomed wurden und welche nicht.

Nibbels, 2017-07-29:
Punkt 2 wurde inzwischen in der development behoben.
Die anderen Vorschläge sind relativ zu den Änderungen nicht mehr so dringend oder kollidieren evtl. mit der gewünschten Freiheit eines Bedieners.

Ich schließe das.

closed 2017-07-29 by Nibbels