RF1000community/Repetier-Firmware #12: Vorschlag: mhier-Scan-Hilfe beim Einstellen der korrekten Höhe der Z-Schraube wäre gut.

Nibbels, 2017-03-14:
Es scheint, als wären viele Drucker extrem verstellt, was die Z-Schraube angeht. Die meisten Nutzer wissen das nicht und sind sich über die Auswirkungen nicht im Klaren.

Man könnte einen Selbsttest machen, der durch den mhier-Scan und die Abfrage der aktuellen Hotend-Temperatur und Bett-Temperatur dem Nutzer mitteilt, ob seine Z-Schraube ideal eingestellt ist oder gravierend verstellt ist.

Am Ende könnte man am Display anzeigen, ob das OK ist oder nicht.

Vorgehen: 
- Temperatur abfragen und beim Hotend ca. +1um pro °C annehmen.
- Temperatur abfragen und beim Heizbett ca. -1.5um pro °C Grad annehmen
- (Durchgewärmt vs. Kalt sind nur ca. 150um unterschied, das allein könnte man tolerieren)
- Hinweis auf saubere Düse geben, Play-Knoipf verlangen.
- mhier-Scan durchführen und Ergebnis verrechnen.
- Solleinstellung: Heißes Hotend bei -0.3mm, d.h. Heißes durchgewärmtes Hotend max. bei ~-0.1
- Anhand der Steigung der Z-Schraube ausgeben, wieviele Umdrehungen eine Korrektur nach oben oder unten erfordert.

Nibbels, 2017-03-14:
Völlig klar ist mir dabei auch, dass die Z-Schraube nicht zwingend das Problem sein muss.
Generell sollte man in der Firmware eine Variable haben, die aussagt, wie weit der aktuell verbaute Schalter überfahren werden darf.
Z_OVERRIDE_MAX

Diese Einstellung sollte eine sinnvolle Auswirkung auf HEAT_BED_Z_COMPENSATION_MAX_MM HEAT_BED_Z_COMPENSATION_MIN_MM Z_OVERRIDE_MAX und andere Begrenzer haben, die fest codiert sind.
Z.B. Suchlimits beim mhier-Scan oder HBS.

mhier, 2017-03-14:
Solange die Z-Kompensation nicht Bit-Überläufe bekommt, ist es mit dem Z-Scan vollkommen egal, wo die Schraube steht. Selbst positive Kompensations-Offsets funktionieren meines Wissens problemlos. Beim Standard RF1000-Endschalter würde der Z-Schalter kaputtgehen bevor Bit-Überläufe passieren, demnach sollte das alles kein Problem sein...

Nibbels, 2017-03-14:
Ich hatte mal gehört, dass die Z-Matrix eigentlich immer negativ sein sollte. Und daran hab ich mich anschließend gehalten, weil ich nichts riskieren wollte.

Der Z-Scan bricht z.B. aktuell ab, wenn er was positives produzieren würde.
Dabei müsste man evtl. mal überdenken, ob ich in diesem Punkt nicht zu restriktiv war.

mhier, 2017-03-15:
Vielleicht täusche ich mich in dem Punkt, müsste man prüfen.

Nibbels, 2017-03-15:
Zumindest ein Z-Homing von über einer positiven Matrix-Position kracht theoretisch in die Platte.

mhier, 2017-03-15:
Das stimmt. Ich sehe ja nach wie vor keinen Sinn darin, einzelne Achsen zu homen...

Wessix, 2017-03-18:
doch mache ich ab und zu. Ob das andere jetzt als sinnvoll erachten sei mal dahingestellt. aber manchmal erledige ich z.b. Reinigungsarbeiten und rekonditionierungsarbeiten bei heißem Druckbett. Dazu fahre ich auf halbe höhe und bewege nur in Y Richtung. Wenn ich dann jedesmal ganz hoch fahren müsste wäre das extrem nervig ...

Nibbels, 2017-03-19:
Diese neue Funktion ist nun am Ende dem Z-Offset-Scan eingebunden:
https://github.com/Nibbels/Repetier-Firmware/blob/community_development/Repetier/RF.cpp#L2170

> 0:31:17.505: ZOS(): STEP 9 : GOTO z=0
0:31:19.663: X:80.00 Y:90.00 Z:-0.87 E:0.00
0:31:19.759: ZOS(): finished (2)

> 0:31:19.759: Z-Schrauben-Helper:
0:31:19.801: - Alle Werte in Mikrometern / Einheit [um] -
0:31:19.802: Matrix-Minimum: -796.48
0:31:19.802: Weitere Extruderausdehnung maximal: 253.05
0:31:19.802: Weitere Heizbettausdehnung maximal: 146.77
0:31:19.802: Minimalabstand bei Maximaltemperaturen: -396.66
0:31:19.803: Wenn das Druckergehaeuse gerade durchgewaermt waere: -396.66
0:31:19.803: Wenn das Druckergehaeuse gerade kalt waere: -546.66
0:31:19.803: #############
0:31:19.803: Sollkorrektur: 197 [um] = 0.197 [mm]
0:31:19.803: Sollumdrehungen: 0.4 (weiter reindrehen)
0:31:19.804: (RF2000: Minimal eine halbe Schraubendrehung einstellbar.)
0:31:19.804: #############

Nach der Korrektur:
(Erneuter Z-Offset-Scan)

> 0:37:08.258: Z-Schrauben-Helper:
0:37:08.258: - Alle Werte in Mikrometern / Einheit [um] -
0:37:08.258: Matrix-Minimum: -544.92
0:37:08.258: Weitere Extruderausdehnung maximal: 252.85
0:37:08.259: Weitere Heizbettausdehnung maximal: 146.62
0:37:08.259: Minimalabstand bei Maximaltemperaturen: -145.45
0:37:08.259: Wenn das Druckergehaeuse gerade durchgewaermt waere: -145.45
0:37:08.259: Wenn das Druckergehaeuse gerade kalt waere: -295.45
0:37:08.260: #############
0:37:08.260: Sollkorrektur: -55 [um] = -0.055 [mm]
0:37:08.260: Sollumdrehungen: -0.1 (RF2000: Die Z-Schraube ist ok!)
0:37:08.274: (RF2000: Minimal eine halbe Schraubendrehung einstellbar.)
0:37:08.274: #############

💃  🗡 :D

closed 2017-03-19 by Nibbels