RF1000community/Repetier-Firmware #13: Vorschlag: Motorstromoptimierer / Accelerationtuning

Nibbels, 2017-03-14:
Man könnte die Beschleunigungseinstellungen anhand der Motor-Strom-Vorgabe autotrimmen.
Oder die Motor-Ströme anhand der Beschleunigungseinstellungen autotrimmen.

Vorgehen:
- Homen.
- Voll anlauf in X,Y,Z
- Vollbremsung in X,Y,Z
- Wiederholtes Wiederholen von 2 und 3.
- Positionsschritte bis zum Erreichen des Endschalters zurückzählen.
- Sehen, ob Steps übersprungen wurden.

Auswerten: 
- Motor-Ströme absenken und Test Wiederholen
- Bei Fail Motorstöme mit Sicherheitsfaktor einstellen.
- Drucker-Rating in der Log ausgeben

Auswerten: 
- Beschleunigung erhöhen
- Bei Fail Beschleunigung mit Sicherheitsfaktor einstellen.
- Drucker-Rating in der Log ausgeben.

Warum??

- Der Drucker könnte leiser sein.
- Die Motoren könnten kühler sein.
- Der Drucker könnte schneller drucken.
- Durch diesen Test wäre die Druckermechanik vergleichbar. Man könnte z.B. direkt Marcometaner mitteilen, dass der jeweilige Drucker extrem schlechte Werte in Y aufweist, während X und Z ok scheinen...
- Ebenso könnte man die Effekte von Umbauten oder Schmierungen der Linearachsen zahlenmäßig einordnen.

mhier, 2017-03-14:
Vielleicht sollte man ein paar passende G/M-Codes einführen und die eigentliche Logik in einem kleinen Python-Tool implementieren. Das alles in die Firmware zu prügeln, ist vielleicht etwas viel und auch zu unflexibel.

Nibbels, 2017-03-14:
Ja...
Von meiner Seite aus bleibt das erstmal ne Idee, die ich nicht vergessen will.

Praktisch wäre das sicher, aber ich bin beim ersten Anlauf nicht 100% mit den ganzen Bewegungsfunktionen in der Firmware zurechtgekommen und hatte es anschließend auf Eis gelegt.

Nibbels, 2017-03-19:
Vorher in jedem Fall https://github.com/RF1000community/Repetier-Firmware/issues/7

Nibbels, 2017-08-07:
Ganz interessante Lektüre, wenn es um die Auswirkung der Beschleunigungseinstellung geht:
https://github.com/repetier/Repetier-Firmware/wiki/Hardware-settings-and-print-quality#needed-force

Ebenso das Thema
https://github.com/repetier/Repetier-Firmware/wiki/Hardware-settings-and-print-quality#extruder-pressure-control (USE_ADVANCE)
Dort gehts um die Kompensation des Materialflussversatzes bei Beschleunigungen, durch die Federwirkung/den Druckaufbau des Filaments im Hotend.

Nibbels, 2017-08-07:
Änderung der Motorströme ist während des Betriebes ganz einfach möglich!
https://github.com/Nibbels/Repetier-Firmware/commit/274e1684cf62befd512acd687405ef04ac88f18b

Nibbels, 2017-09-01:
Durch einen Test mit alternativen Spindeln ...
... und etwas Erfahrung.

Wenn man im Menü die Motorströme einfach einstellen kann ist die Autofunktion nicht möglich. Man verfährt und wenn man verlorene Steps bemerkt macht man die Ströme etwas höher.
Ausserdem ist z.B. die Einstellung des Z-Stroms, wenn man die Spindeln verspannt festschraubt sehr viel anders als wenn man sie korrekt verschraubt. Ein Tuning auf das letzte Prozent wäre irrsinn.
Für 10% bis 20% Spiel ist Autotune nicht nötig.

 Ich schließe das Thema.

closed 2017-09-01 by Nibbels