RF1000community/Repetier-Firmware #16: RF2000/RF1000: HEAT_BED_Z_COMPENSATION_MIN_MM Einigung auf Wert.

Nibbels, 2017-03-14:
@mhier  

https://github.com/RF1000community/Repetier-Firmware/blob/community_development/Repetier/RF1000.h#L956
#define HEAT_BED_Z_COMPENSATION_MIN_MM          float(0.5)    
vs.
https://github.com/RF1000community/Repetier-Firmware/blob/community_development/Repetier/RF2000.h#L945
#define HEAT_BED_Z_COMPENSATION_MIN_MM          float(0.2)        

betrifft: 
long			g_minZCompensationSteps	   = HEAT_BED_Z_COMPENSATION_MIN_STEPS;

Und wird vom ebenfalls von SenseOffset als obere Grenze verwendet, g_minZCompensationSteps sagt aus, wann das SenseOffset aufhört zu arbeiten.
SenseOffset kann mitlaufen, wenn die Digits in den höheren Lagen nicht größer sind als in der ersten Lage. Das sollte nicht der Fall sein.

Kritisch ist ist das nicht, aber wir sollten uns evtl. auf eine Strategie einigen, was in der Konfiguration voreingestellt ist.

mhier, 2017-03-14:
Im Original ist der Wert in beiden Fällen 0.2. Da ich keinen RF2000 habe, hab ich den Wert wohl nicht geändert. Hast du ihn auf 0.5 erhöht?

mhier, 2017-03-14:
Ggf. muss man eine zusätzliche Grenze einführen, bis wohin der SenseOffset läuft. Solange es keinen Grund gibt, dass beide Werte identisch sind, sind getrennte Konfigurations-Parameter sinnvoll.

Nibbels, 2017-03-14:
Ok, woher das kommt, müsste man recherchieren. HEAT_BED_Z_COMPENSATION_MIN_MM ist in der Development 1.37 ebenfalls auf 0.2mm eingestellt. 
(Es war beim RF1000 so eingestellt, nicht beim RF2000.)
Ich habs nun vereinheitlicht!
Wenn du es nicht gezielt so haben wolltest, ist es wohl egal.

Generell könnte man beide Limits trennen, nur im Normalfall ist es nicht wichtig. Wer seinen ersten Layer über Z=+0.2mm, braucht SenseOffset normalerweise nicht.
Und dieser könnte auch auf die Idee kommen, HEAT_BED_Z_COMPENSATION_MIN_MM per MCode etwas höher zu stellen.

closed 2017-03-14 by Nibbels

Nibbels, 2017-03-19:
Hrmpf...
https://github.com/RF1000community/Repetier-Firmware/issues/21