RF1000community/Repetier-Firmware #18: Vorschlag: Geschwindigkeitsdrosselung oder Auto-Pause bei extrem hohen Digits (Limit einstellbar)

Nibbels, 2017-03-17:
Ähnlich SenseOffset, nur sollte das die Geschwindigkeit runterregeln und notfalls in ner Pause landen.
Wenn das Hotend ausfällt oder die Düse zukokelt wäre das ein ganz praktisches Feature.
(Verhindert Ritzeldurchdrehen oder Extruderklackern)

Die Temperatur müsste im Fall einer Pause beim Extruder runter, beim Hotend nicht.

Wessix, 2017-03-18:
und einen Versuch in eine "frei" position zu fahren eine und dan versuchen nach etwas zeit ganz vorsichtig etwas zu extrudieren und zu schauen ob die digit werte dann fallen. quasi freispülen im leeren raum.

Nibbels, 2017-05-19:
**Die Auto-Pause gibts schon.**

Das ist die Emergency-Pause.
Die kann man aufbohren, in dem man zweimal "pausePrint();" statt einmal Pauseprint einfügt.

Anschließend kann man M3075 S-xxxxx Pxxxxx nutzen um die Max-Digits einzustellen.
Klappt gut, aber der Bug https://github.com/RF1000community/Repetier-Firmware/issues/34 hängt da mit drin.

RF1000 weiß bescheid und hat ein Bugvideo.
Ich warte gespannt...

Nibbels, 2017-08-04:
Das hier wäre besonders für PETG gut.
Allerdings funktioniert weniger Geschwindigkeit hier nicht :) In dem Fall braucht man bei hohen Digits meist einen Pull oder etwas Oiling.

Da es Auto-Pause schon gibt, schließe ich diese Idee.

LG

closed 2017-08-04 by Nibbels