RF1000community/Repetier-Firmware #19: hliebscher Bug mit Anzeige

Nibbels, 2017-03-17:
"wenn der Digit wert über 10.000 geht ,macht der Drucker kurz Geräusche und im Display verschwindet die Anzeige"

Ich weiß, dass ich beim RF2000 ein paar mal eine leere Anzeige hatte. Aber bisher habe ich das der statischen Entladung am USB-Masse-Kabel zugeschrieben.

Nibbels, 2017-03-17:
Tests: Ich konnte es bisher beim RF2000 nicht nachstellen!
Darum habe ich weitere Infos angefragt:

-10000 und +10000 per Daumendrücken ohne dass er druckt ist kein Problem.
SenseOffset aktiv ohne Drucken ... +-10000 macht nichts aus.
SenseOffset Aktiv im Hauptmenü +-10000 macht nichts aus.
+-10000 Bei Z=0 und aktiver SenseOffset ohne zu drucken macht nichts aus.
Bei +-10000 während dem druck der startmade passiert auch nichts
Immer egal ob ich im Haupt oder Mod-Menü bin.
Während dem Druck passiert auch nichts, wenn ich über die Grenzen +-10000 gehe.

Fragen:

"Beim RF1000 ist das Display 16 Zeichen lang.
Kann es sein, dass ein 5-Stelliger-Digitwert nicht ins Menü passt?"

"Nur scheint das Abstürzen bei mir Manuell nicht zu klappen. Bist du länger und dauerhaft über 10000?"

"Tritt das im Hauptmenü auf? Oder nur im neuen Mod-Menü?`"

Wessix, 2017-03-17:
Also 5 stellige Digit Werte gehen

Nibbels, 2017-03-18:
Danke!

Ich hab meine Firmware in englischer Sprache und dort hatte ich den Bildschirmausfall inzwischen mehrmals. Meistens während dem Drucken, der druckt weiter.

Nibbels, 2017-03-19:
**Das liegt an der statischen Entladung.**
auch hliebscher hat das nun bestätigt. Das Display fiel dann aus, als er mit der Spachtel ans Heizbett kam.
Vermutlich wegen der Induktion.

Da ist was nicht korrekt geerdet.
Das war auch im Forum schon angemerkt worden.

Es trat bei mir am RF2000 auch auf.
 (Ikeastuhl Markus + Pulli + Drucker)
Und bei Wessix am RF1000 war mal die LED-Leiste Ursache für eine ähnliche Systematik, nur mit mehr Stromfluss.

closed 2017-03-19 by Nibbels