RF1000community/Repetier-Firmware #20: Z-Position Menü: Knopfrichtung vs. Bettrichtung.

Nibbels, 2017-03-18:
Will man das Bett nach unten verfahren drückt man intuitiv die Knöpfe nach unten.
Ausser man ist schon dran gewohnt.

Mathematisch macht man damit das Increment negativ, also fährt das Bett nach oben.

Darf ich das umstellen?
Meinungen?

Fix: ui.cpp Zeile 2833
https://github.com/RF1000community/Repetier-Firmware/blob/community_development/Repetier/ui.cpp#L2833
`
        case UI_ACTION_ZPOSITION:
        {
            nextPreviousZAction( increment );`

zu `
        case UI_ACTION_ZPOSITION:
        {
            nextPreviousZAction( -1*increment );`

Nibbels, 2017-03-18:
Für mich: Getestet und absolut für gut empfunden.

Nibbels, 2017-03-19:
Ich habs geändert!
https://github.com/Nibbels/Repetier-Firmware/commit/d260f2890440679852daad24d9278edc2cca1830

Vorsicht an die Fräser!

mhier, 2017-03-19:
Ich find das nicht gut, das ist doch eindeutig falsch!?

Nibbels, 2017-03-19:
Wie würdest du das erwarten?

- Knopf nach unten und Heizbett fährt nach oben?
- Knopf nach unten und Heizbett fährt nach unten?

Ich hätte ohne Vorwissen ganz klar den zweiten Punkt erwartet.

Nibbels, 2017-03-19:
Ich hab jetzt einfach die Richtung in die configuration.h eingetragen:

#define Z_BUTTON_DIRECTION		    		-1
oder wie bisher:
#define Z_BUTTON_DIRECTION		    		1

Ich meine immer noch dass wir das Bett verfahren und nicht das Werkzeug zum Teil.
Woher kommt diese Anschauung?

mhier, 2017-03-20:
Naja im Menü stellst du eindeutig die Koordinaten ein. Die Z-Koordinaten sind nun mal sorum definiert, dass bei Bewegung in negative Z-Richtung das Bett nach oben fährt. Ich kenne kein Gerät, welches einen Wert erhöht, wenn ich die Taste nach unten drücke, bzw. erniedrigt, wnen ich die nach oben drücke. Das Verhalten mag im ersten Moment nicht ganz intuitiv sein, aber es ergibt sich in eindeutiger Weise aus der Menü-Anzeige. Etwas anderes sind die zusätzlichen Knöpfe, mit denen unbahängig vom Zustand die Z-Achse gefahren werden kann, aber die sind ja auch bereits andersherum.

Genauso könntest du argumentieren, dass bei der X-Achse nicht die Knöpfe nach oben und unten, sondern die Knöpfe nach rechts und links den Druckkopf fahren sollten (welche aber nun mal ganz anders belegt sind). Auch hier gilt, die Tasten haben nichts mit der Bewegungsrichtung zu tun sondern beeinflussen den Koordinaten-Wert.

Vielleicht sollte man lieber (statt dieser Änderung, die für mich klar in die falsche Richtung geht) einen zusätzlichen Menüpunkt einführen, der eine intuitive Bewegung in allen Achsen erlaubt: Die vier Knöpfe des Steuerkreuzes bewegen in X und Y und die ohnehin vorhandenen Zusatztasten für die Z-Achse lassen sich ebenfalls verwenden. Wenn man das gut benennt und einen entsprechenden klaren Hinweis im Display anzeigt, halte ich das für die viel bessere Lösung, als irgendwo die Koordinaten-Richtung entgegen der eigentlich klaren Logik umzudrehen, nur weil das auf den ersten Blick irgendwie intuitiver aussieht.

Nibbels, 2017-03-20:
Sowohl die mathematisch logische und die intuitive Argumentation sind mir völlig vertraut ^^.

**Hinweisschnipsel:**
- Im Repetier-Server ist die Richtung ebenfalls "koordinatenkorrekt verkehrt".
- Wenn man Y-Position bedient, ist das ebenfalls sehr unintuitiv. Taste nach oben: Bett kommt auf einen zu, anstatt wegzufahren.
- _Was für mich generell die Lage entschärft, ist, dass man nicht mehr in den Emergency-Block rasen kann, wenn man die falsche Taste erwischt._
- Was inzwischen ebenfalls positiv ist: Die Z-Sondertasten sind nun völlig funktionsentkoppelt, dort gibts keine Überraschung mehr.
- Meistens wird man fürs alltägliche Bettverfahren die Z-Sondertasten benutzen.
- Es kann gut sein, dass besonders im Fräser-Modus diese Verstellmenüs genutzt werden, darum die Koordinatenachsenanschauung aus diesem Grund besonders wichtig ist.
- Die Idee mit dem Menü finde ich gut, da man aber die Positionsmenüs schon selten nutzt und wegen den Bugfixes mit den Endschaltern glaube ich nicht, dass das viele nutzen würden.

Ich ändere den Standard zurück auf "koordinatenkorrekt". Die bezeichnung Z_BUTTON_DIRECTION	 gefällt mir heute nicht mehr so gut. Da überlege ich mir was besseres und ich mach auch ein kleines Update für die Readme.

LG

mhier, 2017-03-20:
Das Problem an intuitiven Argumentationen ist, dass sie sehr subjektiv sind. Aus meiner Sicht gibt es genau eine korrekte Deutung an der Stelle, auch wenn man diese intuitiv gelegentlich nicht gleich sieht. Ich verwende das Menü genau dann, wenn ich nach Koordinaten gehen will. Das deckt sich damit, dass man dort auch die Distanz einstellen kann, die bei Tastendruck gefahren wird. Wenn ich die Z-Achse um 1mm bewegen möchte, bin ich schon bei Koordinaten.

Ich seh ja auch dann ein, dass man sich da vielleicht mal täuscht und in die falsche Richtung denkt. Aber dann denkt man eben nun mal falsch. Wenn wir das jetzt umdrehen, denken andere (insbesondere ich) noch viel öfter falsch, weil die Tasten einfach nicht mehr das machen, was im Menü geschrieben steht.

Ich halte daher die Funktion für irreführend und würde sie komplett entfernen. Wer das Menü benutzt, wird sich da schnell dran gewöhnen.

Nibbels, 2017-03-20:
> Ich verwende das Menü genau dann, wenn ich nach Koordinaten gehen will.

Das hat mich im Grunde überzeugt. Und weil Anfänger mit dem Mod normalerweise eh nichts zu tun haben.
Einstellbar lasse ichs. Nur dreht die neue Konfigurationskonstante alle Achsen um. Auch Y und sodass es  einheitlich ist auch X. 
```
/**
XYZ_POSITION_BUTTON_DIRECTION = -1 : This fits to you if you want more intuitivity when choosing the Up-Down-Buttons within "Menu:Position->X-/Y-/Z-Position".
XYZ_POSITION_BUTTON_DIRECTION = 1 : This fits more if you want to stick to standard coordinates direction.
*/
#define XYZ_POSITION_BUTTON_DIRECTION		1```

Mir persönlich gefällt das koordinatenverkehrte viel besser, weil ich diese Menüs im Grunde nur nutze wenn ich das Heizbett reinigen will.
Dann muss ich das Bett so hinfahren, wie ich es brauche und die Steppermotoren müssten aktiv sein (wegen dem Wischen)
Genau das und nichts anderes macht für mich dieses das Menü ^^.

mhier, 2017-03-20:
Das macht aber zumindest in X wirklich überhaupt keinen Sinn... ;-) Aber wenn du es brauchst, lass es ruhig drin, solange der Default auf der Original-Einstellung bleibt.

closed 2017-03-20 by Nibbels