RF1000community/Repetier-Firmware #31: SenseOffset Scharfstellung als kleines Symbol im Modmenü zeigen.

Nibbels, 2017-03-28:
Wenn jemand im Slicer eine Höhenveränderung (logisches Z-Offset) einstellt, die allerdings der Drucker nicht kennt, ist manchmal die Schwelle für die SenseOffset-Höhenlimitierung schon überschritten.

Darum wärs für die Schnellhilfe und Kontrolle gut, wenn wir im Display ein kleines Zeichen haben, solange der Drucker denkt, er wäre im ersten Layer.

Wessix, 2017-04-23:
Absolut dafür

mhier, 2017-04-23:
Ich bin ja immer noch eher unglücklich über diese recht undurchsichtigen Automatismen, aber durch eine solche Anzeige wird es zumindest etwas durchsichtiger.

Nibbels, 2017-04-23:
In meiner Testversion ändert sich nun das @ zu nem ^, solange die Regelung aktiv ist.

Beim Test gerade war z.B. bei der Startmade noch "@ 2800" dringestanden und solange er überwacht "^ 2800"
Ist glaub gut!

LG

Nibbels, 2017-04-23:
https://github.com/RF1000community/Repetier-Firmware/commit/82a653195f6b4774521b3523cb6fb7365ba26e3a

Beim nächsten Stable ist das mit drin. Ich teste nun genauer.

mhier, 2017-04-24:
Bitte entsprechend auch in der Dokumentation ergänzen!

Nibbels, 2017-04-24:
Richtig, wollte ich, habs aber vergessen.
Jetzt ist das mit dem Zeichen in der changelog und der Readme erwähnt und ich habe zwei neue Screenshots eingearbeitet.

Nibbels, 2017-04-24:
https://github.com/RF1000community/Repetier-Firmware/commit/1d755c4e43164a99832c639b6a8627169999a0eb

Schaut mal bitte, was fehlen könnte. Dann schließen. :)

mhier, 2017-04-24:
Sieht gut aus, danke!

closed 2017-04-24 by mhier