RF1000community/Repetier-Firmware #32: Startmadenerkennung anhand der Extrusionsmenge pro cm/mm

Nibbels, 2017-03-28:
https://github.com/RF1000community/Repetier-Firmware/issues/31

Zusätzlich zu diesem eben verlinktne Problem, wärs evtl. ziemlich 'gscheid, wenn wir eine globale Variable einführen, in die der Drucker ständig die aktuelle Extrusionsmenge einträgt.
Dann kann man bei überschreiten einer startmadentypischen Minimalmenge SenseOffset ausser Kraft setzen und ansonsten für die ersten vielen Millimeter komplett ohne Nebenwirkungen aktiviert lassen.

mhier, 2017-03-28:
Warum kann man nicht einfach den SenseOffset erst nach der Startmade einschalten? Beides wird doch im Start-GCode vom Benutzer eingetragen.

Nibbels, 2017-03-28:
Richtig... 
aber mit einer Erkennung wäre die Funktion vollautomatisch und quasi idiotensicher. Idiot ist natürlich übertrieben, aber in der Zukunft sollte nicht jeder 3D-Drucker-Bediener alle 200 Gcodes kennen müssen.
Drum die Idee.
Wir haben schon mehrmals kleine Fehlerlein gesucht, die entstanden, weil Simplify3D oder andere Startcodes unglaublich kreativ sind, wie man auf alternativen Wegen ein Offset im Code verbauen könnte.

LG

mhier, 2017-03-28:
Dafür sucht sich dann derjenige dumm und dusselig, bei dem die Erkennung mal fehlschlägt (in einer der beiden Richtungen). Ich fürchte, das wird eher häufiger vorkommen, ist sehr viel schwieriger zu identifizieren und erfordert ein erheblich größeres Wissen, als das bisschen Dokumentation, in welcher Reihenfolge man die Befehle im Startcode braucht. Der SenseOffset muss doch eh im G-Code eingeschaltet werden, d.h. Dokumentation ist eh nötig.

Nibbels, 2017-03-28:
Die Idee entstand, weil 
1) sowohl Wessix wie auch noch jemand anfangs nicht rausfinden konnten, warum SenseOffset nicht hochzählt.
Obwohl aktiviert und obwohl alles vorher funktioniert hatte.
2) man danach die deaktivierungs-Schwelle im Grunde ausbauen könnte, die aktuell an der Z-Kompensation dranhängt.
3) Mindestens mit https://github.com/RF1000community/Repetier-Firmware/issues/31 hätten wir sofort gemerkt, was das Problem war.
4) Manche Leute einfach unglaubliche Layerhöhen drucken, weil sies nicht besser wissen.
5) Man Z-Offset in Hardware einstellen kann und auch Z-Offset in Software umsetzen könnte, worauf ich selbst nie gekommen wäre. Dagegen bin ich bisher absolut Machtlos ^^.

Kann gut sein, dass die Startmadenerkennung auf diese Art übertrieben wäre, aber es wäre vermutlich 1 Zeile Code und eine etwas geänderte Condition beim SenseOffset, drum ist diese Variante evtl nicht schlecht.

LG

mhier, 2017-03-28:
Ich würde trotzdem nicht mindestens zwei neue Fehlerquellen schaffen, um eine los zu werden, die man genausogut durch ohnehin notwendige Dokumentation loswerden kann. Letztlich müsstest du die Schwellwerte dafür auch wieder konfigurierbar machen, am Ende hat das Ding 20 Konfigurationsparameter, die kein Mensch mehr überblickt. Dann funktioniert das noch genau für die Entwickler und sonst niemanden mehr, weil für jeden aus irgend einem Grunde einer der Parameter nicht stimmt.

Außerdem verstehe ich bei den meisten deiner Punkte nicht, was die damit zu tun haben :-)

Nibbels, 2017-03-28:
Der geringe Schwellwert, dass SenseOffset ab Layer 1 = max. Z=0.2 / 0.33 inaktiv wird und nicht mehr nachregelt war dazu gedacht, dass man die Startmade bei der SenseOffset-Korrektur automatisch auslässt, weil dabei natürlich höhere Digits entstehen.
Die Made würde sonst ein Offset produzieren, welches schädlich wäre.

Ansonsten könnte SenseOffset im Grunde immer mitlaufen, es bringt aber nur in der ersten Lage was.
Um 1% Fehlerquelle in höheren Lagen auszuschließen bleibts aber gut, wenn es irgendwann sagt:
"Hier kann die erste Lage nicht mehr sein, also wird hier nicht weiter korrigiert, auch wenn das Filament leicht im Hotend klemmt oder zu weich wurde und Digits produziert".

Cool wäre z.B. wenn man diese Max-Grenze auf irgendwas wie 3mm setzen könnte, dann wäre die Fehlerchance insgesamt sicherlich minimal. Nur dann könnte irgendwer auf die Startmade reinfallen.

Schließe ich die Startmade als Offset-Quelle aus, ist der Fall ideal.
Und das kann ich einfach machen, indem ich feststelle:

(Extrudsionslimit < E / SQRT(X*X+Y*Y) ? Normaldruck : Startmade)

LG

Nibbels, 2017-08-04:
Auch das ist nicht die Lösung. Wenn man die Lagenhöhe dünner macht wie die Startmade hoch ist das alles kein Problem. Startmade bei 0.35 und Firstlayer bei 0.25- und es läuft perfekt.

closed 2017-08-04 by Nibbels