RF1000community/Repetier-Firmware #35: Änderung Einleitender Text bezüglich support von mhier

Wessix, 2017-04-23:
Da mhier ja mitlerweile im Boot ist sollte der Satz das diese Version nicht vom Ersteller des Z-Offset Scans unterstützt wird evtl. mit seinem Einverständnis gestrichen werden

mhier, 2017-04-23:
Den Satz hab ich wohl übersehen damals. Bitte gerne entfernen ☺️

Wessix, 2017-04-23:
hab das mal comitted

Nibbels, 2017-04-23:
hehe top

Nibbels, 2017-04-23:
Ok.... jetzt ist das in der Stable geändert, aber in der development nicht. kannst du, mhier, das automatisiert in der development einfügen, ohne die dortigen Anderungen zu überschreiben?

Nibbels, 2017-04-23:
https://github.com/RF1000community/Repetier-Firmware/commit/ccc5c2f323a1f1c02e750a0ad0a204a277d8761f

closed 2017-04-23 by Nibbels

mhier, 2017-04-24:
Am besten nächstes Mal in den development branch committen, dann kommt sowas automatisch beim Release mit (oder man kann es cherry-picken). Eigentlich sollte man in den stable branch nie direkt committen.