RF1000community/Repetier-Firmware #41: PID-Regelung mit M303

Nibbels, 2017-04-26:
Als ich meinen PID-Tune gemacht hatte, stand bei 
Extr. 1 Drive Min = 40
Extr. 1 Drive Max = 40

Der Drucker ist damit nicht so heiß geworden, wie er hätte sein müssen.
Eine änderung auf 0 ... 255 hat das Problem gelöst.
Nun dasselbe bei hliebscher. Auch bei ihm stand dann 40/40 drin.
Er meint, im RF1000 stand vorher 40 und 180 drin. Damit funktioniert wieder alles perfekt.

http://www.rf1000.de/viewtopic.php?f=72&t=1193&p=17986#p17986

Nibbels, 2017-04-26:
Zitat aus http://www.rf1000.de/viewtopic.php?f=67&t=1577&p=15610&hilit=Ziegler#p15610

> mhier https://de.wikipedia.org/wiki/Faustformelverfahren_(Automatisierungstechnik)#Methode_von_Ziegler_und_Nichols

mhier, 2017-04-26:
Wo stehen denn diese falschen Werte (also 40/40)? Die Min und Max Drive-Werte haben ja erstmal nichts mit den PID Parametern zu tun.

Nibbels, 2017-04-26:
Sagen wir so: Genau weiß ichs nicht, nur ist verdächtig, dass mehrere Persionen nach der PID-Erkennung probleme mit Erreichen der Zieltemperatur haben und Drive Min und Drive Max die Lösung zu sein scheint.

Ich kann mir mal die nächsten Tage den zugehörigen Code durchlesen, wenn du aber schon weißt was da lost ist, wärs einfacher :)

Aktuell _scheint_ das Problem erkannt, aber ich habe mich noch nicht weiter drum kümmern können.

mhier, 2017-04-26:
Nein, ich weiß nicht, was da los ist :-) Die Frage war: Wo oder wie stellst du diese Min und Max Werte ein?

Nibbels, 2017-04-26:
Die Werte im EEPROM können in RepetierServer oder auch per GCode geändert werden.
[image]
[image]
[image]

mhier, 2017-04-27:
Hm, ok, das sind offensichtlich doch PID-Parameter, die Beschreibung ist nur falsch. Das ist nicht, wie ich zuerst verstanden hatte, der minimale und maximale Heizstrom, sondern nur der minimale und maximale Anteil des integralen Contollers. Der integrale Controller ist dafür verantwortlich, dass der bei einem rein proportionalen Controllers immer vorhandene Unterschied zwischen Soll- und Ist-Wert auf 0 geregelt wird. Wenn das zu stark begrenzt wird, kann der integrale Controller nichts sinnvolles tun, und damit bleibt die Differenz immer vorhanden, der Extruder ist also immer etwas zu kalt.

Im Source-Code stehen die Default-Werte dafür in der RF1000.h bzw. RF2000.h und heißen dort für das V2-Hotend HT3_PID_INTEGRAL_DRIVE_MAX und HT3_PID_INTEGRAL_DRIVE_MIN. Dort sind die Werte korrekt eingetragen (40 und 180). Wenn ich das richtig überblicke, sind die Werte nicht Teil der Parameter, die automatisch bei der Optimierung angepasst werden. Die falschen Werte müssen also irgendwie anders ins EEPROM gekommen sein. Die Variable im Source-Code heißt "pidDriveMax", ich habe den kompletten Source-Code danach durchsucht, es gibt nur wenige Treffer.

Nibbels, 2017-04-27:
> M303 P10 S250 X0

`23:55:17.168: Info:PID Autotune start
23:57:26.374: bias: 74 d: 74 min: 247.14 max: 258.33
23:57:49.115: bias: 73 d: 73 min: 246.67 max: 253.33
23:58:11.776: bias: 72 d: 72 min: 246.67 max: 253.33
23:58:11.776: Ku: 13.75 Tu: 22.64
23:58:11.777: Classic PID
23:58:11.777: Kp: 8.25
23:58:11.777: Ki: 0.73
23:58:11.777: Kd: 23.35
23:58:34.619: bias: 72 d: 72 min: 246.67 max: 253.33
23:58:34.634: Ku: 13.75 Tu: 22.86
23:58:34.635: Classic PID
23:58:34.635: Kp: 8.25
23:58:34.635: Ki: 0.72
23:58:34.635: Kd: 23.58
23:58:57.460: bias: 72 d: 72 min: 246.67 max: 253.33
23:58:57.461: Ku: 13.75 Tu: 22.82
23:58:57.461: Classic PID
23:58:57.461: Kp: 8.25
23:58:57.461: Ki: 0.72
23:58:57.461: Kd: 23.53
23:58:57.462: Info:PID Autotune finished ! Place the Kp, Ki and Kd constants in the Configuration.h or EEPROM`

Anschließend war drive max und drive min ok = unverändert.

mhier, 2017-04-28:
Dann liegt es wohl nicht an der Firmware, zumindest nicht an der aktuellen Version. Vielleicht war früher mal ein Bug in der Firmware, dann können ja falsche Werte im EEPROM stehen bleiben.

Ich würde vorschlagen, das Ticket erstmal zu schließen.

Nibbels, 2017-05-03:
ok

closed 2017-05-03 by Nibbels

Nibbels, 2017-07-31:
Darum war das immer 40 / 40
https://github.com/RF1000community/Repetier-Firmware/commit/91b9303876709a6c2ea3d6f258d4359244bc6670

Habs in der Stable gepached, weils ein Bug ist.
In der Development ist das auch drin, ich committe das später von meinem Account, wenn ich alles getestet habe.