RF1000community/Repetier-Firmware #42: Idee: Digits-Offset zur Nullung der Digits.

Nibbels, 2017-04-26:
Mag nicht jeder für absolut Sinnvoll halten, aber cool wärs sicherlich.

- Mcode abschicken
- Ermittelung der Ruhe-Digits
- Alle zukünftigen Digits werden intern um die ermittelten Ruhe-Digits korrigiert.

Allerdings wird diese Idee vermutlich etwas länger hier stehen bleiben, weil es nicht zwingend notwendig ist und man wegen Überläufen etc. alles prüfen sollte.

Wie kam ich drauf? Ich habe umgebaut und nun andere Digits wie bisher. Nun müsste ich mich umgewöhnen oder überall +1000 rechnen. Mit der Normalisierung müsste ich das nie wieder. Dann wären Digits nur noch rein Filamentabhängig / Verschmutzungsabhängig

mhier, 2017-04-27:
Damit versteckst du aber nur wahrscheinliche Probleme. Wenn die Ruhe-Digits zu groß werden, ist da mechanisch was verspannt. Das führt zu größeren Ungenauigkeiten bei allen Messungen, die die Digits verwenden, und damit z.B. zu Abbrüchen beim HBS. Ich würde dir eher empfehlen, einfach mal alle beteiligten Schrauben zu lockern und reihum in kleinen Schritten wieder anzuziehen. Damit solltest du die Digits wieder runter bringen. Einzige Ausnahme ist, wenn du tatsächlich das Gewicht des Extruders verändert hast. Aber wie oft kommt das vor?

Ich finde, wir sollten die Firmware nicht überladen mit Gimmicks, die in irgendwelchen seltenen Randbereichen vielleicht mal ausnahmsweise sinnvoll sein können. Erstens muss irgendwer den Überblick behalten und z.B. Bugs noch zuordnen, finden und beseitigen können. Den Überblick hat glaube ich jetzt schon keiner mehr, und das wird immer schwieriger. Zweitens führt das dazu, dass die Bedienung irgendwann für alle, die die Entwicklung nicht mitverfolgt haben (also für alle außer uns), unmöglich wird. Es ist irgendwann einfach nicht mehr klar, für welches Problem welches Feature benutzt werden soll. Dann fangen die Leute an, irgendwas zu machen, und die Probleme verschwinden nur scheibar, dafür tauchen dann neue auf. Durch Dokumentation lässt sich das nicht lösen, schau dir mal an wie lang schon jetzt unser README ist. Das liest doch kein Mensch mehr :-)

Nibbels, 2017-04-27:
Ja, wegen Featureüberladung bin ich völlig deiner Meinung.

Diese Idee war dadurch motiviert, um einen Standard zu schaffen.
"Digit-Homing" per G-Code. :D
Der Funktion für den SenseOffset würde das helfen! Anschließend kann man auf seine Anweisungen im Start-Code des Slicers besser vertrauen.
Wie gesagt: Ich lasse die Idee etwas liegen. Wenn mir noch ein Grund einfällt, denke ich drüber nach. Aktuell steht sowieso Drucker-Umbau an.

mhier, 2017-04-27:
Dann lös das Problem doch im SenseOffset durch eine Leermessung. Wie gesagt, es wird dazu führen, dass der HBS Probleme macht und mit Sicherheit auch der Z-Offset-Scan. Evtl. funktioniert auch der SenseOffset nicht mehr so gut. Die Verspannungen führen zu stärkeren (und vor allem nicht statistischen) Schwankungen der Digits und machen die Messung unzuverlässig. Das springt dann gerne mal in einen anderen Zustand, weil sich die Verspannung teilweise löst o.ä.

Nibbels, 2017-04-27:
Die Leermessung bei SenseOffset finde ich nicht gut, weil man dann nicht mehr so einfach ablesen kann. Das würde SenseOffset selbst komplizierter machen und nicht nur ein Homing darstellen. Der Bediener sieht dann ständig z.B. F=2345, hat eine Leerlaufzahl von -1000. Müsste dann aber ca. 3300 einstellen und nicht 2300.

Ich habe über den Vorschlag damals länger nachgedacht und kam zu diesem Entschluss.

mhier, 2017-04-27:
IMO ist ein "Homing" hier aber schlicht falsch. Zumindest solange die Dehnungsmessstreifen nicht defekt sind, entspricht ein Wert von F=0 einer Kraft von 0 auf den Dehnungsmessstreifen. Da der Extruder etwas wiegt, gibt es einen konstanten Offset, der der Gewichtskraft des Extruders entspricht. Jede Abweichung davon wird durch eine zusätzliche Kraft erzeugt. Wenn du eine zusätzliche Kraft im Leerlauf hast, solltest du die Kraft tatsächlich beseitigen und nicht einfach wegrechnen. Die einzigen "legitimen" Anwendungen des von dir vorgeschlagenen "Homings" wären also eine Änderung des Extrudergewichts (also z.B. Umbau auf einen anderen Extruder) oder die Korrektur von defekten Dehnungsmessstreifen (wobei hier ein Austausch doch sehr empfehlenswert wäre). Erst wenn es dir auf Genauigkeiten von wenigen Digits ankommt, macht ein Abnullen des Wertes Sinn, aber dann auch nur unmittelbar vor der Messung (wird so z.B. beim HBS gemacht).

Ich würde also vorschlagen, den Wert bewusst nicht zu korrigieren sondern die Nutzer lieber dazu zu motivieren, das eigentliche Problem zu beseitigen.

Nibbels, 2017-07-16:
Ich habs für mich und Harry eingebaut: das ist in der Praxis absolut super!
Homen und anschließend ließt man F: -1 .. 0 ... 1
Es ist einfach vergleichbarer als sich zu merken, ob man bei -1000 oder 1000 gestartet hat. Vorallem, wenn man Sense-Offset oder Digit-Kompensation nutzt.

Siehe FEATURE_ZERO_DIGITS in configuration.h
Branche development.

closed 2017-07-16 by Nibbels