RF1000community/Repetier-Firmware #44: Drive-Free

Nibbels, 2017-04-27:
Wieder gabs Ärger damit.
Wir sollten das rausschmeißen...

mhier, 2017-04-27:
Von mir gerne :-)

Nibbels, 2017-04-27:
Weißt du genau, wo man das Skalpell ansetzen müsste?

Ich hab das noch nie live gesehen und könnte zwar irgendwas abklemmen, aber nichts testen.

mhier, 2017-04-27:
Ich kann mir das demnächst mal ansehen, aber wohl erst nächste Woche...

Nibbels, 2017-05-03:
Vermutlich würde es reichen, wenn wir bei
https://github.com/RF1000community/Repetier-Firmware/blob/community_stable/Repetier/RF.cpp#L7246

die Stelle `char    driveFree = 1;` auf 0 setzen, dann wird es niemals aktiv.

Wenn Freifahren aus anderen Gründen ein Problem sein könnte, (weil es irgendwelche Z-Fahr-Verbote mit oder ohne Homing, Z-Compensation oder sonstwas geben könnte) können wir die Funktion `freeZ( -Printer::axisStepsPerMM[Z_AXIS] );` auch auf einen Menüpunkt legen.

Nibbels, 2017-05-19:
Soll ich das killen, ich weiß inzwischen wie?

Nur testen ob der millingmode/printermode irgendwie doch davon abhängig ist, kann ich leider nicht.
(Testen ob es dann einen Zustand gäbe, in dem Z durch keinen Trick verfahrbar wäre... etc.)

mhier, 2017-05-22:
Sorry, ich komm im Moment nicht dazu. Wenn du weißt wie, mach es gerne und ich teste es demnächst mal. Sonst mach ich es irgendwann später :-)

Nibbels, 2017-07-16:
In der Development ist das nun raus. Ob das Auswirkungen auf den Betrieb der RF1000 hat, kann ich selbst nicht beantworten.

mhier, 2017-07-18:
Ich habe gestern einen ersten Druck mit der neuesten Version gemacht und schon mal keine nervigen "Drive Free"s mehr beobachtet. Ich hatte trotzdem aber noch einmal den Bug, dass das Z-Homing nicht korrekt funktioniert hatte. Ich muss das noch mal genauer analysieren.

Nibbels, 2017-07-18:
Wenn du was rausfindest, aber nicht gleich weißt, wo es im Code schiefgeht, sags mir :)
Ich schau da parallel gerne drüber. 

Beim RF2000 scheint das nichts auszumachen. Mein Drucker funktioniert perfekt. Notfalls müssten wir mal jemand anders überreden das zu installieren, um einzugrenzen ob es nur im Modus Circuit fehlerhaft ist.

Nibbels, 2017-07-29:
Wenn z.B. mit der neuen Version das Z-Homing besser läuft können wir das schließen.
- Funktioniert das Z-Homing nicht, wenn einer der Knöpfe gedrückt ist?
- Funktioniert in diesem Fall noch die Bedienung mit den Z-Tasten?
- Braucht der Drucker an einer bestimmten Stelle eine Fehlermeldung auf dem Display?

mhier, 2017-08-21:
Mir ist gerade aufgefallen, dass jetzt, wenn der Z-Schalter als "Circuit" konfiguriert ist und der Drucker bei gedrücktem Endschalter aufwacht, beim ersten Homing das Bett direkt nach oben fährt. Das würde den Endschalter zerstören, wenn er am oberen Endschalter steht. Hängt das evtl. zusammen mit dieser Änderung? Ich finde leider den Commit gerade nicht, um es mal testweise zu reverten...

mhier, 2017-08-21:
Ok, Problem behoben, siehe fb9207e51d411e11f53e725b53e2db6f8a502510

Nibbels, 2017-08-21:
Das Richtungs-Problem, dass wenn der Drucker mit gedrücktem Schalter aufwacht war die Ursache für den Bug mit Drive-Free.

Ich sehe du hast ein paar Commits, darum warte ich erstmal ab. Danke!!

Um den richtigen aktuell gedrückten Endschalter herauszufinden, würde ich eine Art "Sinus in Z" anlegen der linear größer wird. Abbrechen, sobald der Endschalter losgelassen wurde. Dann wäre die Erkennung absolut zerstörungsfrei und zuverlässig, wenn der Knopf nicht mehr als halb durchgedrückt war.

mhier, 2017-08-22:
Im Prinzip kann man immer gefahrlos nach unten fahren, weil der untere Z-Schalter das ab kann. Das ist offiziell so von Conrad vorgesehen. Mit einem Sinus in Z würde man gefahr laufen, den oberen Schalter zu weit zu überfahren und ggf. sogar mit dem Extruder ins Heizbett zu fahren, wenn der sich gerade über dem Bett befindet. Für mich hat es jetzt gut funktioniert bei ersten Trocken-Tests, mal sehen, ob es auch zuverlässig läuft. Ich hab in nächster Zeit noch Einiges zu drucken. Ich lass das Ticket hier erstmal offen.

mhier, 2017-09-10:
Für mich hat das jetzt gut funktioniert, sowohl beim Fräsen als auch beim Drucken. Der Fehler, dass bei mir das Z-Homing beim Drucken gelegentlich schief geht, weil er zu früh wieder nach oben fährt (bevor der Z-Schalter losgelassen wurde), habe ich aber immer noch, wenn der Z-Schalter auf Circuit konfiguriert ist. Das ist wohl ein Problem unabhängig von driving free...

Nibbels, 2017-09-10:
Kannst du das genauer erklären?

Der fährt im Druckmodus von unten nach oben, kommt an, fährt etwas weiter  runter und weider etwas langsamer nach oben. Dort bleibt er zum Schalterkontakt stehen.

So sollte es sein, oder?

Es gibt in der RF1000.h/RF2000.h soeine Einstellung:
```
/** \brief If during homing the endstop is reached, how many mm should the printer move back for the second try */
#define ENDSTOP_X_BACK_MOVE                 5
#define ENDSTOP_Y_BACK_MOVE                 5
#define ENDSTOP_Z_BACK_MOVE                 0.5
```
Ist es möglich, dass dieser Backmove was damit zu tun hat?

Nibbels, 2017-10-18:
AtlonXP hat im Printmodus ziemlich alles durchgetestet:
Er hat nur einen "Single"-Schalter am RF1000 und kann nun eine Z-Matrix ertasten, die im Bereich bis -12mm liegt.

Homing funktioniert immer, auch von dieser Tiefe heraus.

Das Extra-#define für Z_ENDSTOP_DRIVE_OVER und die Abhängigkeit der anderen Werte davon ( https://github.com/RF1000community/Repetier-Firmware/commit/2704206ead75fa777c6e5c1503b00f9a8ae15272#diff-42d04a53302d660aad46837650f4ff9dR633 )
ist ne gute Sache :)