RF1000community/Repetier-Firmware #52: Default-Werte für manual steps in X und Y zu groß

mhier, 2017-07-17:
Ich würde gerne die Default-Werte für die manual steps in X und Y (DEFAULT_MANUAL_STEPS_X und DEFAULT_MANUAL_STEPS_Y) verringern z.B. auf 1/20mm. Aktuell ist die Größe ca. 1/2mm, was für viele Zwecke viel zu groß ist und letztlich überflüssig, da es ja bereits 1mm als Option im Menü gibt. Spricht etwas dagegen, wenn ich das committe?

Nibbels, 2017-07-17:
Z kann man einstellen.
X und Y habe ich bisher nie beachtet, weil die der normale Druck-Nutzer nicht braucht.

**Stell das um :)** 
Ich ziehe mir dann für meine Offline-Version deine Änderung neu und patche das unter Umständen in den anderen Branches.

LG

mhier, 2017-07-25:
Erledigt! (vergessen zu pushen... ;-) )

closed 2017-07-25 by mhier