RF1000community/Repetier-Firmware #81: Find Z Origin funktioniert nicht mehr

mhier, 2017-09-10:
Wenn ich im Fräs-Modus Find-Z-Origin ausführe, hängt er am Ende des Scans fest in der Zeile "GCode::executeFString( Com::tFindZOrigin );". Dort wird ein Script aus der RF1000.h Konfiguration ausgeführt, welches nach Z=15mm fährt. Wenn ich die Zeile kommentiere und hinterher von Hand die selben Befehle ausführe, funktioniert es problemlos. An der Stelle wurde nichts geändert gegenüber der Original-Firmware, früher hat dies aber funktoniert. Ich überblicke leider gerade nicht, wo es hakt...

Nibbels, 2017-09-10:
Habs getestet und es tut nichts.

In jedem Fall wird ein GCode über GCode::executeFString() ausgeführt.
 innerhalb findZOrigin
 innerhalb loopRF
 innerhalb checkForPeriodicalActions

Dasselbe passiert über if( g_uBlockSDCommands > 1 ) eigentlich auch wenn man OutputObject auslöst.

Für den Fall, dass ich 3 byte zuviel für statische Variablen abgeknöpft habe -> 
Test: #define MOVE_CACHE_SIZE                     16
auf #define MOVE_CACHE_SIZE                     15
und testen.
Ich schau mal, ob ichs morgen hinbekomme.

LG

Nibbels, 2017-09-11:
ok.... sonst träume ich davon:
https://github.com/RF1000community/Repetier-Firmware/commit/986a246b5faf625357e3202a93b7e8cb38916ce1

Da war mal was. Wir hatten diskutiert ob G90 G91 manchmal nicht zündet.
Dann hatte ich zur Sicherheit diese Waitings eingebaut.
http://www.rf1000.de/viewtopic.php?f=7&t=1343&hilit=G91&start=50#p18585

Aber die sind natürlich Mist, wenn man das von intern abfeuert.
Hat eben wieder funktioniert.

LG

Nibbels, 2017-09-11:
Fix in 1.37u2
https://github.com/Nibbels/Repetier-Firmware/commit/9c0e9277fc25714a3b45a5742fcb0c37a4430fc7

Kannst du mir den Effekt bestätigen und schließen?

mhier, 2017-09-11:
Danke! Teste ich sobald ich kann, ist allerdings erst nächste Woche :-)

mhier, 2017-10-02:
Gestern getestet: funktioniert wieder!

closed 2017-10-02 by mhier