RF1000community/Repetier-Firmware #132: [Issue Split] Z-Offset-Scan Einstellproblematik

Nibbels, 2018-03-11:
Alter Thread: https://github.com/RF1000community/Repetier-Firmware/issues/129

Hier weiter wegen dem Z-Offset-Scan.
Natürlich können wir das Verhalten ändern!

Warum wurde das geändert:
Weil ich mit PeterKA fast verzweifelt ausschließen wollte, warum sein Z-Offsetscan bei ihm keine sehr guten Ergebnisse gebracht hat. Und wenn man sich die Korrekturwerte angesehen hat, stand am Ende immer sowas wie "dZ=256" oder "dZ=512"
Das war verdächtig.

Ich habe also überlegt, wie das mit den Schritten funktioniert.
Ich fahre im Beispiel schrittweise in z.B. 0.01er Schritten ans Bett ran.
Merke ich, dass die Digits größer werden bin ich mit hoher Wahrscheinlichkeit schon über den Punkt rausgeschossen, den ich eigentlich erfassen würde.
Fehler: 0 bis 0.099999

Eigentlich müsste ich in diesem ersten Überschreitungspunkt einen vollen Interval zurückfahren und feiner annähern.
Das haben wir gemacht.

Zusätzlich: Wenn ich mit einem Interval angenähert habe, fahre ich zurück bis die Kraft wieder abgenommen hat. Nicht zu weit, denn wenn die DMS-Streifen oder z.B. eingelegtes ABS gerade entspannen laufen mir grundsätzlich die Digits weg. Es kann also sein, dass die Kraft die ich als Aufsetzpunkt gemessen habe nur entspannendes Filament war.
Also muss die Kraft beim Zurücksetzen innerhalb eines Intervals wieder abnehmen, sonst ist es nicht die Oberfläche.
-> Log: 10 auf 7 und dann werden die Digits neu genullt.

Ich selbst hatte keine Probleme mit PLA diesen Scan zu machen. Auch nicht, wenn das PLA unten raussabbert und ich in der Suppe messe.

**Wenn du sagst, der geht nicht mit genügend Druck aufs Bett um das Material wegzudrücken, habe ich darum diese zwei Tipps:**
- Dein Material ist noch zu fest.
- Die Limits in der Annäherungs-Trigger-Kraft sind einfach zu klein gewählt.

Wenn du sagst, der Scan dauert zu lange, dann könnte ich mir vorstellen:
- Nach jedem Aufsetzen und Aufbauen einer Kraft will also der Scan die Kraft wieder abbauen. Das ist nur dann blöd, wenn in dem Fall die Intervallschrittlein so winzig sind, dass es ewig zu dauern scheint und genau dann die Digits eben leicht weggelaufen sind?
Ich könnte mir vorstellen, dass wir für diese Fälle auch die Schritt-Zeit verkürzen könnten. Das ist einfach nur ne Delay-Zahl, keine echte Geschwindigkeit.
https://github.com/RF1000community/Repetier-Firmware/blob/4b176392023c1b1ce510fe1d7877c5a95911dbc9/Repetier/RF.cpp#L5247
Oder weniger fein ranfahren:
https://github.com/RF1000community/Repetier-Firmware/blob/4b176392023c1b1ce510fe1d7877c5a95911dbc9/Repetier/RF1000.h#L1095
(Diese Iterations xteln immer die Schrittweiten)

Ich werd mir weiter Gedanken machen, aber zu allererst ist mir eigentlich das Homing-Problme wichtiger, denn das ist echt dubios. Hier sinds glaube ich nur Einstellprobleme.

LG

Nibbels, 2018-03-11:
Die Anpressdrücke haben wir anscheinend nicht geändert.
Git Blame liefert:
https://github.com/RF1000community/Repetier-Firmware/commit/908c85535c3ec765aaac9c34bdd2a882a6d0bdb7#diff-42d04a53302d660aad46837650f4ff9dL1045

Kannst du es bei dir aber trotzdem mit höheren Anpess-Limits versuchen?
SEARCH_HEAT_BED_OFFSET_CONTACT_PRESSURE_DELTA   100 ?

mhier, 2018-03-14:
Ach ich steig nicht mehr durch was der Scan jetzt eigentlich alles tut... Irgendwie verhält sich der Scan einfach nicht so, wie ich das erwarten würde. Bei eingelegtem Filament und heißer Düse funktioniert es meiner Erfahrung nach nicht, nur weniger Schritte vom Kontaktpunkt zurück zu fahren und dann langsam neu zu scannen. Man braucht schon zwischendurch einen deutlichen Abstand, bevor man tatsächlich eine unabhängige zweite Messung starten kann. Dadurch dass der Scan jetzt nach dem ersten Versuch sofort in einen ultra-langsam und ultra-vorsichtig Modus schaltet, kommt er letztlich nie wieder vom Heizbett weg und misst einfach nur Blödsinn. Ich glaube, bei der aktuellen Implementierung ist der komplette Teil ab dem ersten Kontakt mehr oder weniger wirklungslos.

Der Grund ist, dass die Düse dazu neigt, durch das immer vorhandene Filament am Heizbett "festzukleben". Wenn man nicht wieder so weit vom Heizbett wegfährt, dass diese "Verklebung" gelöst wird, haben die Digits wenig damit zu tun, was wir messen wollen. Da hilft auch kein höherer contact pressure - dann passiert das selbe einfach nur bei etwas kleinerem Abstand und damit höherem Druck.

Wenn wir den Scan bei aufgeheizter Düse unterstützen wollen, müssen wir die Logik irgendwie überarbeiten.

Wie benutzt du den Scan genau? Benutzt du ihn nur im kalten Zustand?

Evtl. hängen die Unterschiede auch noch von der Beschaffenheit des Heizbettes und des Materials ab. Der RF1000 und der RF2000 haben unterschiedliche Heizbetten, an denen evtl. das Material in diesem Zusammenhang unterschiedlich gut haftet. Ich habe dann noch dazu eine Dauerdruckplatte, damit verhält es sich wieder anders - und vor allem habe ich ein bisschen Bedenken, den Scan zu lange laufen zu lassen (die Platte kann Temperaturen über 120 Grad eigentlich nicht ab).

Nibbels, 2018-03-15:
Der Z-Offset-Scan funktioniert so wie er ist bei mir hochgenau und weiderholbar. Aber ich nutze Pertinax oder Glaskeramik. Tatsächlich nichts, was kleben kann. Eher immer "trockene" Oberflächen, wenn das Pertinax auch keine Dauerhitze mag.

Ich erkläre mal kurz die Idee:

Wir fahren bei 0 Digits los.
Und zwar in großen Schritten bis wir -40 digits überschritten haben.
Vermutlich landen wir aber drüber. Darum fahre ich zurück zum letzten Interval und dann feiner nochmal ran.
Der Z-Wert wird immer dort gemessen, wo beim Rückfahren die Kraft wieder etwas abgenommen hat.

Aktuell macht der 2 Iterationen. Bei der zweiten wird feiner gemessen, aber die erste könnte reichen.

Lange zurückfahrten wollte ich vermeiden, weil das so ewig dauert, wenn man die Schritte vorwärts so fein macht.
Was ich bei der moveZ-Funktion nicht nicht getestet habe, ist, sie bei kleinen Schrittlein gleichviel schneller laufen zu lassen.

#define XYZ_STEPPER_HIGH_DELAY              100                                                 // [us]
#define XYZ_STEPPER_LOW_DELAY               100                                                 // [us]

Theoretisch schaffen wir also nicht mehr als 5000 Schritte pro Sekunde bei 2560 Schritten pro Sekunde: 
2mm/s.
Das ist aber viel langsamer, weil noch andere Timeouts reinspielen.

Funktioniert das generell bei dir, wenn du 
`#define SEARCH_HEAT_BED_OFFSET_SCAN_ITERATIONS          2`
auf 
`#define SEARCH_HEAT_BED_OFFSET_SCAN_ITERATIONS          1 
`
änderst?

Nibbels, 2018-03-15:
Ok, du kannst das mit 
#define SEARCH_HEAT_BED_OFFSET_SCAN_ITERATIONS 1
noch nicht testen, das funktioniert aktuell leider nicht ^^.

LG

Nibbels, 2018-03-15:
Auf mehr Iterationen kann man glaube ich verzichten. Stimmt!
Die Ergebnisse werden gleich bis ähnlich. Ich habe ~ 0.02 Unterschied.
Ich mache nun eine Version fertig, die einfach schneller ist und sage dann Bescheid.

Nibbels, 2018-03-15:
Ok!

Teste bitte mal die https://github.com/Nibbels/Repetier-Firmware/ 1.41.09 +
Der hat bei mir sogar ein raushängendes fast gekühltes Filament korrigiert und nur +20 Steps höher gemessen als sonst.

Wir fahren das Bett ja vor dem Messen sowieso an. Da ist die Iteration nicht mehr so wichtig wie damals, glaube ich.

LG

mhier, 2018-03-15:
Nee, mit Filament drin kann ich nicht auf die Iterationen verzichten. Bei den ersten Iterationen wird erstmal das Material weggedrückt und erst die späteren dann messen den tatsächlichen Abstand. Deswegen hatte ich die Iterationen doch damals eingeführt. Ich brauche letzlich mehrere *gleichartige* Iterationen, von denen am Ende das Minimum genommen wird (also kleinster Abstand).

Nibbels, 2018-03-15:
Das Erstanfahren des Bettes ist was anderes wie die Messung anschließend.
Wenn er anfängt zu messen, hat er schon begriffen, dass er auf der Höhe des Bettes ist und mindestens 1x reingedrückt.

Hier hab ich ein Video mit sauberer Düse:
https://youtu.be/fMQkmQcvRWk

Dann hab ich nochmal eins gemacht, to der sich echt vortasten muss. Nur leider hab ich natürlich im Video immer falsch geraten, wann er sich nach untengetastet hatte.
Natürlich ganz am Schluss und nicht wenn ichs sag :D
https://youtu.be/jup5gKlgsok
Man merkt, dass es komplizierter zu sein scheint und länger dauert, wenn mehr Dreck und Filament an der Düse hängt, aber der Scan müsste eigentlich robust sein.

Ich habe nun die Restgenauigkeit bei der Messung am Ende weggenommen. Das war unnötig und vermutlich einfach ein Resultat aus unserer Z-Paranoidität wegen PeterKAs Probleme.

Schau dir das mal an und teste das mal!

mhier, 2018-03-16:
Das blöde ist, dass wir vermutlich auch sensitiv sind auf Änderungen der absoluten Skala, selbst wenn die Wiederholgenauigkeit hoch ist/bleibt. Es kann also durchaus sein, dass ich für die neue Firmware auch wieder funktionierende Einstellungen finde, nur sind diese eben anderes als mit der alten Firmware. Das sollte man vielleicht im Hinterkopf behalten, wenn in Zukunft solche Änderungen vorgenommen werden. Mein Vorschlag wäre, dass du mir einaml möglichst exakt beschreibst, wie du den Scan benutzt und was du in deinem Start-Code machst. Dann probiere ich mal aus, ob ich damit auch reproduzierbare Ergebnisse bekomme.

Du scheinst ein E3D v6 Hotend zu haben, wenn ich das auf dem Video richtig erkenne. Das kann durchaus auch noch Einfluss haben. Ich habe inzwischen beides (Original v2 und E3D v6) wechselseitig im Betrieb und bin gerade dabei, auf Dual-Extruder umzustellen. Dabei würfelt sich eh alles durcheinander, aber vielleicht ist das eine gute Gelegenheit, die Robustheit der Algorithmen in sehr unterschiedlichen Kombinationen auszuprobieren (2 verschiedene Hotends mit 2 verschiedenen Druck-Oberflächen und 2 verschiedenen Materialien - theoretisch kriege ich also 8 Konfigurationen hin).

Btw: wie funktionieren die ganzen Scans eigentlich mit zwei Extrudern? Hängt das da nicht von der Oberflächenstruktur ab, welcher der beiden Extruder zuerst Kontakt mit der Platte hat?

Nibbels, 2018-03-16:
Genau die Kombinationen sind das Problem! Du bekommst aber mehr als 8 hin.
DIe Vielzahl der Hotends und Betten ist noch kombiniert 
- mit Popeln/Degradation auf der Scanstelle, 
- ob man Filament drin hat oder nicht. 
- welcher Filamenttyp eingelegt ist. 
- ob man gerade noch ein kaltes Hotend hatte oder dieses bereits warm war.

ABS zieht sich von warm nach kalt zusammen und wenn sich das Hotend erstmals aufwärmt, laufen u.U. die Digits weg.
Das war der erste Grund, den Scan leicht zu ändern. Dann hatte ich das Problem mit den 2^x-Schritten in den Ergebnissen, dann haben wir den Grund 100% festmachen können, warum bei PeterKA etwas anderes Verhalten war und ich habs mit der Präzision am Ende übertrieben, weil ich das ausschließen wollte 🗡 
Ich glaube, ich habe Wochen damit verbracht, dieses ABS-Problem zu lösen und die Präzision wieder herzustellen. Unser Trick: Der Druck auf eine solide Oberfläche geht zurück, wenn man zurückfährt. Der Druck in Matsch oder der Druck durch weggelaufene Digits machen das nicht.

Mir gefällt der Scan nun wieder viel besser. Manchmal muss man etwas Abstand gewinnen und neu rangehen. Doch wir müssten das im Alltag testen.

> Btw: wie funktionieren die ganzen Scans eigentlich mit zwei Extrudern? Hängt das da nicht von der Oberflächenstruktur ab, welcher der beiden Extruder zuerst Kontakt mit der Platte hat?

Das heiße Hotend kommt zuerst auf. Das wird vermessen. Ausser das Bett ist extrem uneben.
Wenn dein Hotend nach unten mit Anschlag gefedert gelagert wäre (+-0.5mm zum linken), könntes du im Mod dieses Tipdown-Offset aktivieren. Dann fährt dir das Bett je nach Hotend in die richtige Höhe. Von der Programmierung her sollte alles passen, aber ich habe das Drucken damit nicht durchgezogen - Mein Dual-Tipdown-Hotend dann wieder ausgebaut. Und drucke nun Single E3D :D

> Das sollte man vielleicht im Hinterkopf behalten, wenn in Zukunft solche Änderungen vorgenommen werden. Mein Vorschlag wäre, dass du mir einaml möglichst exakt beschreibst, wie du den Scan benutzt [...]

Ich hoffe, daran zukünftig nichts mehr ändern zu müssen. Wenn du sagst, das geht mit deiner Platte und dann die anderen noch funktionieren, sollte alles in Ordnung sein.

@Wessix wäre der Richtige, um das genaue Vorgehen zu diskutieren. Er arbeitet vor jedem Druck mit dem M3900.

Ich drucke unabhängig vom Startcode mit einem vorjustierten Drucker.
Mein Startcode sieht genau so aus:

```
[Simplify macht die Temperaturen am Anfang und wartet drauf]
G28 ; home all axes
G92 E0 ; Filamentwegreset
M3001 ; activate Z-Compensation 
M3909 ; activate SensiblePressure
M3912 S100 P4500 I2 ; sensible Startmade mit 2 Linien
```

Das Einstellen:
- HBS-Scan kalt (normalerweise/nur zu Testzwecken Scan PLA)
- Heizen und ohne Nachdehnung M3900
- Dann speichern der Matrix mit M3902 Sx
- Die Matrix ist also so eingestellt, dass sie etwas näher dran ist. Das korrigiert beim Brim unser SenseOffset.

Das nachjustieren für eine andere Düse/Temperatur:
- M3900
Das Nachjustieren, wenn was am Bett gemacht wurde:
- M3900 R1 oder Menü->Configuration->Z-Calibration->Auto Matr. Leveling
Dann immer Abspeichern der Höhe mit M3902 Sx

Das da: 
`M3912 S100 P4500 I2`
Ist eine Startmade, die auf Druck langsamer wird und weniger extrudiert, also die Digits nicht so übertreibt. Das kann ein paar Probleme beim Druckbeginn beheben. Vorallem mit dem E3D/Fullmetal Hotend. https://youtu.be/0cXA-UMHLNU

LG

Nibbels, 2018-04-01:
^^
Quelle: http://www.rf1000.de/viewtopic.php?f=67&t=2084&start=70#p23098
mhier sagt
> Ich bin gestern auch umgestiegen auf die 01.41.11, allerdings habe ich das Homing mit gedrücktem Schalter und der Circuit-Einstellung noch nicht getestet. Aber immerhin scheint jetzt der Z-Offset-Scan und der Sense-Offset wieder zuverlässig zu arbeiten. Für den Z-Offset-Scan versuche ich jetzt gar nicht mehr erst die Düse sauber zu bekommen und lasse das Hotend dafür auf Druck-Temperatur. Getestet habe ich bisher aber nur mit ABS, E3Dv6 und RF1000-Original-Keramik-Oberfläche. Ich bin mir noch unsicher, ob der Scan nicht für meine Dauerdruckplatte zu Lange dauert (die benutze ich nur für PLA)...
Schließ den Thread einfach, wenn man ihn schließen kann.

closed 2018-04-13 by Nibbels