RF1000community/Repetier-Firmware #200: SD card reader - print stops

fbeck55, 2019-12-21:
Hallo Zusammen,

Ich habe einen RF2000, und seit letzter Woche mit Nibbels/Repetier_Firmware development drauf.

Drucke normalerweise ab SD card, dort kommt im moment beim compile der Fehler:
> #error SD printing is currently not working very well. It works but we saw are sporadic drives to x and y home or extrusion problems.
> #error Remove this warning if you know what you are doing and if you really want to print from SD with all possible risks.

Habe es trotzdem ausprobiert und der Print stand dann tatsächlich nach 8 Layern plötzlich still. Heizer wurden abgestellt, es wurde aber nicht die Standard-Routine mit dem Freifahren des Teils gestartet.

Ich möchte das weiter beobachten. Habt ihr da schon gewisse Erkenntnisse, welche ihr mit mir teilen könnt, was wurde geändert was möglicherweise Einfluss auf das Drucken ab SD hat?

Danke & Gruss FB

Nibbels, 2019-12-26:
Wir hatten die FW konstant weiterentwickelt. Und irgendwann fiel auf der Drucker spinnt. 
Aber das fiel nicht sofort auf, weil ich SD nie nutze. 

Wir haben irgendwann empirisch rausgefunden, dass das nur auftritt wenn man von SD druckt. Spontane 0 zwischenfahrten und so. Das ist bei XY nervig aber tolerierbar. In Z aber nicht.

Ich hatte alles abgesucht und mir monate den kopf zermartert. Aber es lag 99% nicht an der core firmware. Evtl.am Interrupttiming... Aber der usb datentransfer kann CRC.

Ich hatte mal eine neuere SD lib eingebaut. Evtl. frisst die anders am minimal freien ram. Ich tippe drauf man müsste die auf die 2013 version zurückbauen. Sodass sie mit dem alten chip zuverlässig funktioniert. Aber auch dann will ich das nicht wirklich testen und validieren ^^. 

Mit SD werde ich nicht mehr drucken. Raspberrypi und octoprint/repetierserver sind zu gut.
Ausserdem spart das weglassen des sd moduls ram. Der dient dann als bewegungspuffer. Das bringt was wenn man winzige Pfadstücke wie in kreisen drucken will ohne ins Throttling zu laufen.

LG

fbeck55, 2019-12-27:
Vielleicht macht es in diesem Fall mehr Sinn das SD-printing ganz sterben zu lassen.

Danke für die Auskünfte.
LG FB

Nibbels, 2019-12-27:
Das mit der SD-Funktion wurmt mich natürlich. ^^
Nur habe ich das Thema nach geschätzt einem Monat Arbeitszeit einfach seingelassen und deaktiviert.
Natürlich hätte ich gerne eine funktionierende SD-Variante - nur um die Firmware abschließend rund zu machen.

Was mich nur wundert ist, dass die original Repetier Firmware V1 heute auch diese SD-Library nutzt. Ich habe das Update damals quasi nur "nachgemacht" und in die Conrad-Edition der Repetier-V1 rübergezogen.

Evtl. werde ich in der nächsten Zeit mal die neueren Commits ansehen, die Repetier inzwischen gemacht hat und die für uns interessanten Teile rüberholen.
https://github.com/repetier/Repetier-Firmware/commits/development?after=878d984eed15cdef313c2b09fe7686e24c9afc3d+34
Das ist aber mit dem zeitlichen Abstand nicht sehr einfach, weil Conrad unsere RF-Firmware ca. 2014 weggebranched und stark verändert hat, sich anschließend aber nie wieder um die Bugfixes des original Repositorys gekümmert hat. Ich habe die Community-Firmware nun auch grob ein Jahr liegengelassen und nichts mehr gemacht.
Das habe dann quasi ich übernommen, weil die alte FW echt nicht gut lief. Und im Grunde gingen wir Conrad dann solange mit Bugfixes auf den Sack, dass die sich nun garnicht mehr melden :D 
Naja wird wohl andere Gründe haben :P 

LG

fbeck55, 2019-12-27:
Das versteh ich natürlich. Es ist nur so, ich komm nicht gerade von der programmiererseite. Kann also selbst höchstens etwas debugging auf einfachem nieau betreiben. Dazu bin ich auch weiterhin gerne bereit, sollte es doch mal wieder thema werden.
LG

Nibbels, 2020-01-03:
Servus ^^.

Ich konnte das nicht so stehen lassen und habe nochmal einen Anlauf unternommen.
Zuerst habe ich ein paar TODOs erledigt, die ich mir übers Jahr gesammelt habe.
Dann habe ich den kompletten SD-Support ausgebaut. -> 1.44.00

Anschließend nur genau diese SD-Ausbau-Änderungen revidiert und neben die alten Änderungen gelegt, mit denen ich die SD-Lib ausgetauscht hatte.
Gestern nacht habe ich alles abgeglichen, die neue SD-Lib gelöscht und anschließend durch das Original ersetzt.
Reini aus dem Forum testet die für mich, jedoch sind mehrere Tester durchaus ratsam, weil der Bug wohl sporadisch auftaucht.
Könnest du dir die neue Version auch ansehen?

Das wäre Stand jetzt die 1.45.00 und die findest du vorerst nur hier: 
https://github.com/Nibbels/Repetier-Firmware

LG

fbeck55, 2020-01-05:
Top! Danke dir.
Werde ich machen, gebe dann Bescheid.

fbeck55, 2020-01-07:
Ich habe mit Version 1.45.0.0 einen 6h Druck ab SD machen können ohne Abbruch.
Die ersten 3h konnte ich keine Home-moves beobachten, danach liess ich den Drucker arbeiten^^

Ich werde diese Version ab sofort nutzen und werde allfällige Beobachtungen melden, bis hierhin sieht's aber gut aus! :) 
Sag mir wenn ich noch was Spezifisches testen soll. 
Hab gesehen, dass in der Firmware ein RAM-Überwachungsfeature vorhanden wäre, soll ich das mal für einen Print aktivieren?

Danke dir!

Nibbels, 2020-01-07:
Schön :)
und gerne.

Das klingt schonmal gut. Ich hab auch ne zweite Meinung eingeholt und derjenige meint ebenfalls die FW wäre aktuell in Ordnung. Aber man muss vorsichtig sein, wenn man zufällige Fehler zu früh ok-meldet.

Sollte dir noch was auffallen, melde das hier dringendst ^^. Ansonsten warte ich noch ein paar Meinungen ab und übernehme dann die Änderungen in die Community-Development und Community-Stable. Dann sage ich auch im Forum bescheid.

LG

Nibbels, 2020-01-23:
Wir haben ein paar Tester gefunden. Da diese keinerlei Probleme mit der originalen SD-Karte hatten, habe ich die Änderungen auf den stable_branch gemerged ;)

closed 2020-01-23 by Nibbels