RF1000community/Repetier-Firmware #205: Fräsen deaktiviert

mhier, 2020-03-15:
Der Fräs-Modus ist aktuell ja deaktiviert. Ich habe ihn zum Testen aktiviert und keinerlei Probleme festgestellt. Allerdings habe ich bisher nur ohne Z-Kompensation gefräst. Der Oberflächenscan funktioniert aber. (Ich habe nur vergessen, die Z-Kompensation einzuschalten - war für den Zweck aber auch nicht wichtig.)

Aus meiner Sicht kann man das eigentlich wieder aktivieren.

@Nibbels Gab es konkrete Gründe bzw. Vermutungen, warum du dich entschieden hattest, den Fräs-Modus zu deaktivieren? Oder war das nur allgemeine Vorsicht?

Nibbels, 2020-03-15:
Ich hab damals das komplette Koordinatensystem von int-Steps auf Float umgestellt.
Um zu wissen obs was bringt und wegen der Rundungsproblematik/Summierproblematik bei Pfadstückchen die <1 step beinhalteten. (0-Steps-Pfade ergaben immer Vollbremsungen, Stockeln).
Dabei flog einiges an wirrem RF-Code raus, weil ich der Meinung war, dass z.B. sinngemäß Code-Duplication offensichtlich ist, aber das doch anders und mehrfach programmiert war.
Zudem habe ich dann die Offsets im Display nach rechts gepackt, wenn man die Koordinaten anschaut.

Fürs default Deaktivieren gabs schlussendlich mehrere Gründe:

**Nebengründe:**
- Es nutzt kaum einer den Fräsmodus und es gab immer wieder Leute die gemeint haben der Drucker ist kaputt / das Homing spinnt. Aber der Drucker war nur zur Fräse verstellt worden.
- Die Firmware ist für Nur-Drucker damit schlanker und hat ein paar Byte mehr Heap.
- Man muss heute sowieso nach dem Download den Druckertyp einstellen. Dann kann man auch schnell den Milling-Mode aktivieren. Beim Raspberry hatte ich zeitweise nur noch pull gemacht und solange es keine Konflikte gab, war die Firmware immer gleich mit den korrekten Settings für den entsprechenden Drucker kompiliert worden.

**Der gewichtigste Grund war:**
- Die einzelnen Funktionen wie z.B. Z-Kompensation wurden eigentlich getestet, aber nicht 100% und nicht mit dem absoluten Wissen, wie das eigentlich funktionieren müsste. Darum habe ich das per default deaktiviert. Also viele Änderungen und kaum Tester.

mhier, 2020-03-16:
Naja das bedeutet aber auch, dass ich immer selbst compilieren muss, wenn ich den Fräs-Modus haben will, und die Builds von unserem Jenkins nicht taugen. Ich würde das gerne wieder aktivieren.

Nibbels, 2020-03-16:
Wenn du die 1.45.00 drauf hast, ist dein Problem vermutlich gelöst. Es wird von meiner Seite aus erstmal nix mehr passieren. 
Im letzten Jahr hatten wir sozusagen nur "eine Änderung".

Oder ein zweiter milling-stable Branch mit Fräse?

mhier, 2020-03-16:
Dann lieber eine Konfigurations-Option in cmake und in Jenkins dann eine Konfigurations-Matrix...

Ich habe Dinge gerne automatisiert. Alles andere nervt früher oder später nur.

Warum kauft man sich eigentlich so einen Drucker, wenn man damit nicht fräsen will? ;-) Zumindest Sperrholz Fräsen geht prima, die Qualität, die man da auf Anhieb erreicht, schaffst du mit Drucken höchstens nach vielen Versuchen... Ich halte das ja eigentlich für das Kille-Feature von unserem Drucker, deshalb versteh ich nicht so ganz, warum man das deaktivieren will... Die eine Option im Menü kann doch nicht wirklich stören. Nur weil da mal jemand drüber gestolpert ist, finde ich nicht, dass man es deaktivieren sollte. Und der Heap dürfte eigentlich nicht wirklich kleiner werden, oder wirklich nur wenige Bytes, es werden doch eigentlich keine zusätzlichen Variablen angelegt?

Nibbels, 2020-03-16:
Cmake-Schalter wären natürlich sehr sauber. Das kann ich selbst aber nicht effektiv einbauen, das wäre Raten mit Stackoverflow und Google.
Es gibt zudem inzwischen vermehrt RF2000v2 Nutzer, die den Community-Mod verwenden. Um diese Schalter habe ich (glaube ich) damals cmakes erweitert.

Fräsen wäre natürlich toll. Ich habe 2x RF2000v1 als Single-Drucker, aber mir fehlt der Frässatz. Und ich kann in dem Keller wegen Lärm und Staub nicht fräsen. Das ist mir mit meiner Proxxon MF70 schnell aufgefallen. Das kann ich in den nächsten Jahren auch nicht anfangen und darum ist der zusätzliche Code für mich nur Ballast.

Meine Drucker sind auf perfekte erste Layer optimiert und haben inzwischen nur noch ein Filament eingespannt, bis die Spule jeweils leer ist. Das funktioniert auch so gut, dass ich meist nicht mehr nachschaue wie der Druck startet. Ich geh erst in den Keller wenn das Teil fertig ist.

Zum Spielen habe ich den E3D-Toolchanger ;)

mhier, 2020-03-18:
Den cmake-Schalter kann ich wohl einbauen, das ist nicht das Problem. Wenn wir nur am Ende für jedes Feature nen Schalter einbauen, skaliert das nicht :-)

Staubentwicklung ist geringer, als man so denkt, die Einhausung fängt auch einiges ab. Lärm ist nervig im gleichen Raum, aber schon vor der Tür erträglich und im anderen Stockwerk hinter noch einer Tür nicht mehr wirklich zu hören. Zumindest bei Holz ist der Lärm nicht wirklich anders als das Leerlaufgeräusch des Dremels. Das ist eher hochfrequent und damit gut abzuschirmen.

Ich hab für mehrere Geräte einfach keinen Platz :-)

Nibbels, 2020-04-19:
Gibts hier Neuigkeiten?

mhier, 2020-04-20:
Hab erstmal weiter getestet, Bugs gibt es keine (jedenfalls keine neuen). Ich kann demnächst mal nen cmake-Schalter einbauen. Ich würde aber per Defautl das Feature einschalten, schließlich ist das in der Original-Firmware genauso, und ich finde es irgendwie merkwürdig, ein so großes Feature im Mod einfach per Default abzuschalten.

mhier, 2020-05-30:
So, cmake-Schalter ist eingebaut und auf Jenkins entsprechend die Matrix erweitert.

closed 2020-05-30 by mhier