RF1000community/Repetier-Firmware #207: Z-Matrix wird durch M3013 geändert, aber nicht zurück gesetzt

thecky, 2020-04-18:
Hallo,

letztens habe ich etwas mit dem M3013 rumgespielt um mir die Z-Matrixen ausgeben zu lassen. Dabei habe ich festgestellt dass, wenn man eine andere Z-Matrix als die aktive ausgeben lässt, die Z-Matrix verstellt wird (quasi indirekt ein M3009 Sxx).
Ich weiß nicht ob es so gewollt ist, oder ein kleiner Bug ist.

Meiner Meinung nach, sollte folgender Diff dieses abstellen:
https://github.com/RF1000community/Repetier-Firmware/compare/community_development...thecky:community_development

Das ist jetzt einfach mal schnell ins Blaue getippt - getestet habe ich es noch nicht.
Wäre nett, wenn Ihr Euch das einfach mal anschauen könntet.

Danke und Gruß
Thomas

Nibbels, 2020-04-18:
Spitze, das ist mir echt nie aufgefallen ...

Hätte ich meine Drucker nicht gerade beide am Drucken, hätte ich das sofort getestet.
Was reines Codereview angeht siehts meiner Meinung nach gut aus.

(Zwischen Änderung und Wiederherstellen gibts kein Return. Das passiert nur, wenn man eine andere Matrix/Tabelle ausgeben lies. 255 wird als Matrixnummer nie vorkommen. Es ist ein char mit 255. Sogar Kommentare :D. Der Rest ist so chaotisch wie eh und je.)

Du darfst gerne einen Pull-Request erstellen. Dann kann ich den PR mergen, sobald ich das einmal bei meinem Drucker getestet habe.

Gruß Stefan

############################################################
Was mir jetzt noch selbst aufgefallen ist:
[image]
vs.

[image]

Was war mit der 0-ten Matrix ... Legal oder nicht?
Es gibt das Beispiel:

       if (loadCompensationMatrix((unsigned int)(EEPROM_SECTOR_SIZE * g_nActiveHeatBed))) {

Und 0*x = 0 -> 
[image]
0 ist eher nicht gültig. Wenn ich kein S angebe, gibts auch kein pCommand->hasS.
Mit S0 bekäme ich einen Error. Ohne S im Gcode vermutlich nicht -> das kennen wir ja als Standardfall.

      if (nTemp < 0 || nTemp > EEPROM_MAX_HEAT_BED_SECTORS) {

müsste zusätzlich zu deinem Fix tatsächlich

      if (nTemp < 1 || nTemp > EEPROM_MAX_HEAT_BED_SECTORS) {

heißen?  Oder es ist gerade zu spät für mich ...
Ich würde das Limit auf 1 ändern und diese if (Printer::debugErrors()) { überall rausschmeißen, wenn kein Widerspruch kommt.

:)

thecky, 2020-04-19:
Ich habe den PR mal eben erstellt.
Ja, du hast recht - der Test muss auf "nTemp < 1" erfolgen - das habe ich auch gleich einmal mit als Commit reingeschoben.
Das S0 ungültig ist, sieht man auch schön in der saveActiveHeatBedToExtEEPROM Function:
https://github.com/RF1000community/Repetier-Firmware/blob/c0cda316fb83839f31ec99d1d1dcae0fbfd7db26/Repetier/RF.cpp#L5263-L5266
Hier wird auch auf <1 geprüft, ebenso beim M3009 Befehl:
https://github.com/RF1000community/Repetier-Firmware/blob/c0cda316fb83839f31ec99d1d1dcae0fbfd7db26/Repetier/RF.cpp#L6614-L6620
Eventuell kann man den Test auch nach oben verschieben und gleich beim S-Parameter Check mit abfrühstücken, da hier ja nur der PC über den Fehler in Kenntnis gesetzt wird und nicht auf dem Display angezeigt wird. Ich meine an folgende Stelle:
https://github.com/RF1000community/Repetier-Firmware/blob/c0cda316fb83839f31ec99d1d1dcae0fbfd7db26/Repetier/RF.cpp#L6699-L6704

Ob man alle Printer::debugErrors (ich vermute mal Du beziehst das nur auf den M3013) rauswerfen sollte, weiß ich nicht. Wenn dann müsste der Fehler auch auf dem Display angezeigt werden. Ansonsten, bekommt man ja gar kein Feedback.

Nibbels, 2020-04-19:
Daran hab ich jetzt auch net gedacht:
https://www.arduino.cc/reference/en/language/variables/data-types/unsignedchar/
Zumindest das 1.8.12 Arduino will (unsigned char) oder einen cast mit (char)255 wenn man auf den int 255 vergleicht ^^.

[image]

Ich patche das mit der Version auf 127 um.

thecky, 2020-04-19:
Oh, sorry - das ist mein Fehler. Ich hatte nur den Type von nTemp übernommen, aber nicht bedacht dass der signed ist.
Bei meinem avr Projekten nutze ich immer uint8_t und dann ist man wohl "Betriebsblind"...

Nibbels, 2020-04-19:
Kein Problem. Ich habs auch erst nachlesen müssen, weil ich aus der FW quasi ein Jahr raus bin. :D 
Ich korrigiere das wie gesagt und patche noch 2 andere Sachen. Dann erhöhe ich die Version.

Dein Code funktioniert! Habs eben geflashed und getestet. 
Man sollte nur evtl. trotzdem vermeiden den M3013 Sx während dem Drucken auszuführen. Es wird wegem der Z-Comp gut gehen, weil die Kompensationsdaten in dem Fall ohne Unterbrechung 2x umgestellt werden, aber das wird mitten im Druck stark ruckeln und Blobs geben, weil das Laden der Matrix dauert. Sollte noch jemand feststellen, dass bei manchen der RF-Drucker der Watchdog triggert, muss ich nochmal ran. Aber ignorieren wir das mal einfach ^^.

closed 2020-04-19 by Nibbels