RF1000community/Repetier-Firmware #210: Z-Offset-Scan

mhier, 2020-05-31:
Ich verstehe seit kurzem das Ergebnis des Z-Offset-Scans nicht mehr. Erstmal behauptet er immer, "Z-Skrew too low". Die Ziel-Position der Schraube scheint außerhalb des gültigen Bereichs zu liegen, wenn ich so weit drehe wie gefordert, ist der Abstand so groß, dass er nicht mehr bis zum Heizbett kommt.

Außerdem verstehe ich die Zahl  "Screw: XXXmm" nicht. Sie entspricht nicht der Zahl, die ich zuletzt als Z-Offset am Ende des Scans gesehen hab. Letzteres (der Z-Offset am Ende des Scans) scheint hingegen korrekt den Abstand zwischen Z-Schalterposition und Heizbett anzugeben (als negative Zahl). Das Scan-Ergebnis ist eine positive Zahl, die aber kleiner wird, wenn ich die Schraube weiter rausdrehe (d.h. den Abstand Z-Schalterposition zu Heizbett vergrößere).

Was genau soll die Zahl bedeuten? Ist es das Ziel, diese Zahl auf 0 zu bringen? In dem Fall ist sie wohl ein Zielabstand plus dem (negativen) gemessenen Z-Offset. Der Zielabstand ist in dem Fall wohl viel zu groß.

Wäre es nicht logischer, den gemessenen Abstand anzuzeigen?

mhier, 2020-05-31:
Wegen "Ich verstehe seit kurzem..." -> Früher habe ich den Z-Offset-Scan automatisiert in meinem Start-G-Code eingebaut gehabt und nur selten manuell ausgeführt. Neuerdings mache ich das nicht mehr, stattdessen führe ich ihn immer manuell aus, wenn ich den Verdacht habe, es hat sich was am Z-Schalter getan, und speichere die korrigierte Matrix. Ich kann daher nicht so genau sagen, seit wann da so ist. Ich bin mir aber ziemlich sicher, dass das früher mal anders war. Zumindest der Zielabstand war früher mal richtig.

Nibbels, 2020-06-22:
Ich müsste Details evtl. neu nachlesen, denn es ist lange her :)

Sinn war es, dass man in jedem Fall eine negative Matrix hat. Also dass im HomeZ=0=SchalterKontakt die Düse nie das Bett berührt.
Die Werte die du siehst, sagen dir, dir mit "Screw too low" dass du eine positive Matrix hättest.
Schraube zu kurz heißt, das Bett fährt beim Homen zu hoch.

Dann wäre also 0 auf 0 zu hoch gehomed.
Beim RF1000 mit Z-Switch wäre 0.8mm zu weit in den Knopf gefahren wieder falsch. Beim RF2000 1.3mm zu weit gefahren ende. Beim RF2000 mit runtergefeilter Fahne oder beim RF2000v2 kann man 3.3mm ins Minus fahren. (Mit 3.3mm kann man die Schraube nach Gefühl einstellen. Mit 0.8 nicht.)

Um für den pingeligen RF1000 die 0.8 zu erreichen habe ich mal die Extr.-Temperatur * Ausdehnungskoeffizient + Betttemperatur * Ausdehnungskoeffizient + unbekannte Nachdehnung bis 0.14 o.ä. angenommen und verrechnet. Und dann verlässt mich meine Erinnerung.

Der angezeigte Wert sagt in jedem Fall, wieviele Umdrehungen theoretisch in die Zone führen, die passen müsste. Aber ob das noch funktioniert oder sinn macht weiß ich nicht.
Nach meinem Umbau auf das E3D-V6 waren die Werte für mich sowieso eher falsch.

mhier, 2020-06-22:
> Ich müsste Details evtl. neu nachlesen, denn es ist lange her :)

Kein Problem, erstmal geht's mir um das gewünschte Verhalten. Wenn ich das verstanden habe, kann ich ggf. selbst Debuggen.

> Sinn war es, dass man in jedem Fall eine negative Matrix hat. Also dass im
> HomeZ=0=SchalterKontakt die Düse nie das Bett berührt.
> Die Werte die du siehst, sagen dir, dir mit "Screw too low" dass du eine positive Matrix hättest.
> Schraube zu kurz heißt, das Bett fährt beim Homen zu hoch.

Das kann so nicht sein. Ich war definitiv weit vom Bett entfernt. Ich werde sicherheitshalber aber noch mal mir die Matrix ausprinten, dann müsste ich das ja sehen.

> Beim RF1000 mit Z-Switch wäre 0.8mm zu weit in den Knopf gefahren wieder falsch. Beim RF2000 1.3mm zu weit gefahren ende. Beim RF2000 mit runtergefeilter Fahne oder beim RF2000v2 kann man 3.3mm ins Minus fahren. (Mit 3.3mm kann man die Schraube nach Gefühl einstellen. Mit 0.8 nicht.)

Mein RF1000 ist bei weitem nicht so pingelig wie das Original. 3.3mm gehen aber auch nicht. Außerdem begrenzt die FW ihn ja trotzdem auf 0.8mm...

> Um für den pingeligen RF1000 die 0.8 zu erreichen habe ich mal die Extr.-Temperatur * Ausdehnungskoeffizient + Betttemperatur * Ausdehnungskoeffizient + unbekannte Nachdehnung bis 0.14 o.ä. angenommen und verrechnet. Und dann verlässt mich meine Erinnerung.

Heißt das, da werden Korrekturen bei der Messung vorgenommen? Ich dachte, die Idee war, dass die Längenausdehnung mit dem Sense Offset 'online' im Druck korrigiert wird?

> Der angezeigte Wert sagt in jedem Fall, wieviele Umdrehungen theoretisch in die Zone führen, die passen müsste. Aber ob das noch funktioniert oder sinn macht weiß ich nicht.
> Nach meinem Umbau auf das E3D-V6 waren die Werte für mich sowieso eher falsch.

Hm, ich habe auch ein E3D-V6. Ich glaube, da ist aber noch ein anderes Problem. Im Prinzip konnte ich den Scan nicht zufrieden stellen. Entweder kam "Skrew too low" oder man war am Limit mit den 0.8mm.

Mein Vorschlag wäre:

* Die Zahl "Skrew: XXX mm" sollte den direkt gemessenen Abstand zwischen Heizbett und Hotend bei Z=0mm angeben. Damit sollte die Zahl kleiner werden, wenn ich die Schraube reindrehe.
* Wir legen in der Konfigurationsdatei (RFxxxx.h) fest, welches der Zielwert für diese Zahl ist. Damit kann man sich auch für Spezialkonfigurationen behelfen. Per Default würde ich dort die jeweilige ofiziell empfohlene Konfiguration hineinschreiben (RF1000: 0.5mm). Der Wert bezieht sich zumindest beim RF1000 auf den kalten Zustand, somit ist keine Korrektur für das Hotend notwendig.
* Daraus rechnen wir dann unter Annahme einer M3-Schraube (ggf. ebenfalls in RFxxxx.h festzulegende Gewindesteigung - die #defines kosten ja keinen Speicher) aus, wie weit die Schraube gedreht werden muss, um am optimalen Punkt anzukommen.
* Da der Zielwert nur für die Zahl der Schraubenumdrehungen benutzt wird, kann man sich auch ohne Spezialkonfiguration davon abweichende Zielwerte überlegen. Man muss dann lediglich im Kopf die Zahl der Umdrehungen ausrechnen, oder halt ausprobieren.

Was hälst du davon? Für technisch halbwegs versierte Nutzer (wie vermutlich so ziemlich alle Nutzer eines 3D-Druckers) halte ich eine direkte Abstandsangabe für intuitiver als ein korrigierter Wert, insbesondere, wenn die Korrektur falsch sein kann, weil man den Drucker modifiziert hat (wie vermutlich relativ viele Nutzer der Community-Firmware).

Nibbels, 2020-06-22:
Es geh bei meinen Ausführungen rein um die Anzeige des Zahlenwertes. Der Scan selbst macht was er sollte.
Wenn man das Display ignoriert sollte auch alles ok sein.

Das Ganze soll eigentich nur ersetzen, dass jemand mit den 3 Blatt Papier bei 23°C Extruder-Temperatur leveln muss, ohne dass der Drucker vorher angeschaltet gewesen sein darf.
(Oder wars 200°C und 60°C, wenn der Drucker bereits 20Min durchgewärmt ist?)
-> Genau diese ewige Frage wolle ich mit dem Assisstenten erschlagen.

Du siehst auf dem Display nicht einen Abstand sondern die Angabe, wie man die Schraube mit 0.5mm/U(?) Steigung leveln sollte, sodass es ungefähr passt. Ohne sich einen Abzubrechen.

Nibbels, 2020-06-22:
Es geht um die Z-Schraube, nicht um den anschließenden Offset-Scan ansich.

mhier, 2020-06-23:
> Du siehst auf dem Display nicht einen Abstand sondern die Angabe, wie man die Schraube mit 0.5mm/U(?) Steigung leveln sollte, sodass es ungefähr passt. Ohne sich einen Abzubrechen.

Ich sehe irgendwelche Zahlen, die ich nicht verstehe. Ich glaube, da wird zu viel gerechnet. Das ist dann nicht mehr nachvollziehbar. Deswegen mein Vorschlag, die Zahl in mm so zu ändern (mit Änderung des Titels), dass sie eine klar nachvollziehbare Größe darstellt, die ich im Prinzip auch direkt nachmessen könnte. Das ist für mich einzig der Offset der Düse zum Heizbett.

Wenn du noch Minimum und Maximum der aktuellen "Matrix" in die Größe mit einfließen lässt, kommt man in ein Problem, wenn man gerade gar keine gültige Konfiguration (Matrix und Schraubenposition) hat, weil man z.B. etwas umgebaut hat (einmal Fräsmodus und zurück, und nix stimmt mehr). Dann möchte ich einen Z-Scan machen können, bevor ich einen Matrix-Scan mache, um den Matrix-Scan überhaupt durchführen zu können. Das ist zumindest beim RF1000 zwingend notwendig, ebel weil es so wenig Spielraum gibt.

Wenn man nur so indirekte Größen hat, ist das Debugging auch extrem schwer. Wie gesagt, ich war zuletzt (aktuell ist der Drucker im Fräsmodus, da kann ich nicht mel eben testen) in dem Zustand, dass entweder "Screw too low" erschien, oder er den Scan abgebrochen hat. Der errechnete Zielwert für die Schraubenstellung lag also wohl außerhalb der 0.8mm. Mein Bett ist eigentlich nicht sonderlich schief, das habe ich auch extra noch mal geprüft. Ich habe die Meldung am Ende ignoriert und konnte ja trotzdem drucken. Im Prinzip hab ich die Schraube dadurch eingestellt, dass ich auf das Display gestarrt habe und den letzten Z-Wert vor Abschluss des Scans angeschaut habe. Den habe ich dann auf ~-0.3mm eingestellt. Das ist auch genau das, was ich vom Scan erwarten würde. Wenn ich mein Bett nicht gut ausgerichtet habe, dann sehe ich das im Matrix-Scan und muss das eben mechanisch korrigieren. Alles andere ist doch eh mist. Wenn das Bett sehr schief ist, wird das gedruckte Objekt doch genauso schief.

Was spricht denn dagegen, den gemessenen Abstand auszuprinten? Vielleicht kann man anschließend eine zusätzliche Warnung ausprinten, wenn die aktuelle Matrix durch den Scan den Toleranzbereich verlässt. Aber erstmal will ich doch das Scan-Ergebnis selbst sehen und nicht irgend eine zweifach indirekte Interpretation davon.

Nibbels, 2020-06-23:
Da muss ich nochmal drüber nachdenken.

Das Tool hieß ja eigentlich "Schrauben Helper".
Es geht hier nicht um den Abstand Düsenspitze Bett. Wenn ich mich recht erinnere gehts nur um höchstem Matrix-Punkt und dessen Positionierung gegenüber der Z-Schraube.
Die Angaben sollen nur aussagen, ob man die Schraube hoch oder runterdrehen soll. Da ist schon ein klares System dahinter und das steht auch irgendwo. Hast mal die Code-Kommentare durchsucht?

Ein Z-Offset-Scan ganz ohne Matrix kann man glaube ich nicht starten. Es wird am Ende immer der höchste Punkt gesucht und damit das finale Offset gebildet.
Du kannst allerdings um einen initialen Scan zu vermeiden z.B. auch mit einem GCODE eine [0,0,0,0,...;0,0,0,0...] Matrixtabelle festlegen. Nur wenn du dann 0.4mm Welligkeit im Bett hattest bist du nach dem ersten Z-Offset-Scan wieder auf Los.

Die Matrix ist 16bit "hoch". Wir können also nicht mehr als 16k Steps abbilden, dann gehts auch schief. Das sind 12,7mm wen ich mich korrekt erinnere.

Verwechsel das nicht mit dem Z-Offset ansich, denn der sollte eigentlich das Bett exakt leveln. Da bleibt idealisiert kein Offset, obwohl es tatsächlich aber vorhanden bleibt, weil die Scanpunkte der Matrix einen Fehler enthalten. Und damit verglichen wird. Ausserdem bin ich mir nicht ganz sicher wie stark die Spindeln und ihre Lagerung noch einen Einfluss haben, dass man ein paar 1000stel grundsätlich unsicherheit hat.
Ich weiß nur eins: Ich habe den Bug nie gefunden, dass verschiedene Scanpunkte auf dem Bett immer verschiedene Grund-Offsets hinterlassen. Aber vorsicht: Hier gehts wieder um die Endgenauigkeit des Bettes! Die ist immer vorhanden, egal was dir der Schraubenhelfer anzeigt. Der Schraubenhelfer sagt nur aus, ob man zu weit oder weniger weit den Schalter überfährt bevor man dasselbe Ergebnis erreicht: Düse ist bei Zcompensated=0 direkt auf dem Bett, nicht drunter nicht drüber.

Nibbels, 2020-06-23:
> > Vielleicht kann man anschließend eine zusätzliche Warnung ausprinten, wenn die aktuelle Matrix durch den Scan den Toleranzbereich verlässt. Aber erstmal will ich doch das Scan-Ergebnis selbst sehen und nicht irgend eine zweifach indirekte Interpretation davon.

Die Warnung ist "Screw too Low". Das heißt trotzdem dass dein Z-Offset-Scan funktioniert hat.

Das Scan-Ergebnis ist eigentlich immer ok. Es sei denn, der Scan brach ab, weil ein Flex-Nudelchen unten an der Düse hängt o.Ä.

Nur solltest du deinen tiefsten Matrix-Punkt unterhalb von Z-Endstop-Überfahrweg-Max einstellen, dann ist das natürlich doof und die Kompensation hört bei z.B. -0.8 einfach auf.
[Drehe ich die Schraube weiter rein, geht das Bett beim Homing weiter hoch.
Drehe ich die Schraube weiter raus, steht das Bett beim Homing früher still.
Egal welcher Zustand, die Z-Kompensation korrigiert das, unser Bett darf aber trotzdem nicht weiter hochfahren als 0.8mm nach Endschalter-Klick.]

Vielleicht hilft dir das beim Nachdenken oder Code verstehen.
Nur weil ich gerade nicht genau die Details weiß und kaum Zeit habe, heißt das nicht, es ist nicht durchdacht.
Man sollte evtl. die Doku upgraden, die es dazu gibt.

mhier, 2020-06-23:
Wir reden aneinander vorbei :-)

Ja der Scan selbst hat problemlos funktioniert. Ich möchte den Scan aber dazu benutzen, um die Schraube korrekt einzustellen. Dazu taugt er im Moment m.E. nach nicht, denn ich verstehe nicht, was er mir da ausgibt.

[quote]Nur weil ich gerade nicht genau die Details weiß und kaum Zeit habe, heißt das nicht, es ist nicht durchdacht.[/quote]
Deswegen frage ich ja nach :-)

Ich les dann erstmal Code. Vielleicht verstehe ich den jetzt mit den Informationen besser. Ich glaube trotzdem, dass es hilfreicher wäre, den Abstand selbst auszugeben. Vielleicht kriege ich das ja noch irgendwo auf's Display gequetscht...

mhier, 2020-06-23:
M.E. ist da ein Zirkelschluss, wenn ich den Z-Offset-Scan zum Einstellen der Schraube verwende, um einen HBS-Scan korrekt durchführen zu können (also DAVOR). Der Z-Offset-Scan verwendet die HBS-Matrix, um mir die optimale Schraubenposition zu empfehlen. Der HBS-Scan hängt aber ja von einer korrekten Schraubenposition ab. Wenn ich keinen gültigen Startwert habe und beides neu machen muss, habe ich ein Problem. Das funktioniert nur, wenn eines von beidem schon korrekt ist.

Nibbels, 2020-06-24:
Sodele ...

Mal ein paar Real-Fakten:
Ich habe vorher endlich mein Vulcano mit 0.6er Düse eingebaut, die Z-Schraube mit noch einem Abstandshalter verlängert und einen Z-Offsetscan laufen lassen:

[image]
Das heißt übersetzt:
1) Der Scan ist "Ok".
2) Aber schraube den Schraubenkopf 1.531mm weiter runter.
3) Das sind bei der Standard-Gewindesteigung 3.1 Umdrehungen
4) runter. "(Down)"

Nibbels, 2020-06-24:
Den Z-Offsetscan macht man idealerweise wenn der Drucker vorher kalt war, aber gerade eben aufgeheizt wurde.
Dann Matrix speichern.

Mit einem weit überfahrbaren Schalter:
Wenn du willst, dass du ohne Matrix-Scan den Z-Offset-Scan laufen lassen kannst, um mal grob eine Ahnung zu haben, wie man die Z-Schraube stellen sollte, dann könntest du mit https://www.rf1000.de/wiki/index.php/Gcodes#M3902_-_manuelles_Matrix_bearbeiten_und_speichern M3902 E0 eine Null-Matrix-Tabelle in den Ram schreiben.
(Müsste gehen ...) Dann ist es so, als hättest du ein ideal-flaches Bett verbaut und der Z-Offset-Scan müsste funktionieren und dir ganz grobe Schraubentipps geben. Glaube ich.

mhier, 2020-06-26:
> Schraubenkopf runter heißt auch Bett stoppt beim Homing später. Heißt Bett ist näher an der Düse. Heißt Düse ist weiter im Minus. Also stimmen quasi die Vorzeichen, außer du stellst dir das gerade ganz anders vor und hast einen besseren Vorschlag :D

Ne genau so stell ich mir das vor, war bei mir nur immer irgendwie anders.

Ich glaube, ich verstehe langsam, was das Problem ist. Ich hab mal in den Code gescnaut, genauer in `calculateZScrewCorrection()`.

Wenn ich dort alles zusammen addieren, was an Sicherheiten etc. dort aufgeschlagen wird, komme ich auf (275deg-20deg)*0.95um/deg+(120deg-20deg)*1.5um/deg+150um=524.25um, wenn mein Extruder und mein Bett gerade bei 20 Grad sind. Maximal zulässig ist ein Überfahren des Schalters om 800um. Damit darf der höchste und der tiefste Punkt auf dem Bett nicht mehr als ~275um auseinanderliegen. Das ist ganz schön wenig. Die RF1000-Kachel ist ziemlich wellig, schon daher kommen Beiträge in ähnlicher Größenordnung. Und das Bett exakt horizontal auszurichten, ist auch nicht so wahnsinnig einfach. Die einzige Stellschraube, die man hat, ist im Grunde, die beiden Z-Spindeln gegeneinander zu bewegen - aber das ist echt fummelig und hilft auch nur entlang der X-Achse. Vermutlich kann mein Drucker diese Bedingung schlicht nicht erfüllen.

Ich halte diese Vorsicht schlicht für Übertrieben. Beim Drucken spielt das alles überhaupt keine Rolle. Problematisch wäre es nur, wenn man die Z-Achse homed, während die Düse über der höchsten Stelle des Bettes steht. Ich halte es grundsätzlich für problematisch, Z zu homen, wenn die Düse nicht neben dem Bett ist. Der Endschalter kann immer mal versagen - mir ist das schon oft passiert, dass ich an den angestoßen bin und er hat sich verdreht - dann geht die Schraube einfach vorbei. Oder irgendwas hat sich verstellt (Schraube nicht gut festgezogen oder so), und plötzlich liegt Z=0 nen Millimeter weiter oben. IMO, wer über dem Bett (absichtlich) homed und sich dabei das Bett oder die Düse kaputt macht, ist halt selbst schuld. Wenn wir das in Firmware verhindern wollen, müssen wir halt das Z-Homen verbieten, wenn die Düse nicht bekanntermaßen neben dem Bett steht.

Früher hatten wir diesen Sicherheitsaufschlag nicht drin, und ich hatte viel weniger Schwierigkeiten die Schraube zu justieren. So aktuell bringt mir das nix.

Nibbels, 2020-06-26:
Ja, das ist wenig. Genau deshalb habe ich den Schraubenhelfer gebaut gehabt.
Beim RF2000 war mir das sogar zu wenig, da habe ich default 1,3mm statt den 0,8mm.

Ich hoffe, das ist nicht untergegangen: Der Z-Offset-Scan funktioniert unabhängig von den zur Schraubenoptimierung. Du darfst nur nicht zu weit in den Schalter reinfahren. Wenn du "Screw too low" einfach ignorierst und dir vornimmst nie über dem Bett Z zu homen ist alles gut. Dann arbeitet deine Matrix eben im Plus. Z-Kompensation anschalten und der Druck funktioniert.
Oder nicht?

Mich hat das irgendwann so genervt, sodass ich meinen Schalter optimiert habe. Ich kann 3,5mm drüberfahren. In den Schalter rein. Damit kann ich mit bloßem Auge justieren. Und ich mache das "im Minus". Aber das lässt die Firmware nur mit einer veränderten Konstante zu: Z_ENDSTOP_DRIVE_OVER 

Stell deine Config so ein wie du es bei deinem Schalter erlauben kannst:
https://github.com/RF1000community/Repetier-Firmware/blob/37bcd1059a8a05cb9bb04ace86940e8a7c28e810/Repetier/RF1000.h#L36

Diese strikten 0.8mm sind nur für den Standard-Knopf-Schalter.
Was ich allerdings gemacht habe: Z_ENDSTOP_DRIVE_OVER ist heute wirklich als absolutes Limit umgesetzt. Da geht der Drucker nicht mehr drunter. Nicht mit zComp und nicht mit Racecondition. Damit ist der Schalter sicher.

mhier, 2020-06-27:
Ja mir ist klar dass die Korrektur korrekt berechnet wird. Mir geht es nur um den nützlichen Nebeneffekt, damit die Schraube korrekt einzustellen.

Die Matrix ist ja gar nicht positiv. Sie ist nur nicht negativ genug wegen der sehr strikten Sicherheitsaufschläge.

Ich verstehe schlicht nicht, was die Sicherheitsaufschläge bringen sollen. Ich habe den Verdacht, dass sie beim RF1000 gar nicht einhaltbar sind. Vielleicht geht es bei einigen Exemplaren, die zufällig eine besonders ebene Kachel haben, keine Ahnung, vielleicht ist meine Kachel auch besonders rau. Ich würde aber mal nicht von einem Einzelfall ausgehen. Sicherheit ist toll, aber sie soll doch auch nicht einen daran hindern, den Drucker zu benutzen ;-)

Ja ich könnte hingehen und das Z_ENDSTOP_DRIVE_OVER einfach auf was größeres setzen, mein Schalter lässt sich weiter überfahren und würde notfalls auch einfach die Bewegung stoppen (hab ich schon ausprobiert ;-)). Aber irgendwie finde ich das unschön. Ich benutze gerne die Binaries von Jenkins, dafür hab ich's ja aufgesetzt ;-) Außerdem werden vermutlich mehr Leute das Problem haben. Und mit so riesigen Abständen zwischen Z-Schalter und Heizbett habe ich bisher immer schlechte Erfahrung gemacht, die Kompensation scheint dann nicht so gut zu funktionieren und ich bekomme keine gute Haftung. Vielleicht liegt es auch daran, dass ich ein E3D v6 benutze, was meines Wissens eine geringere Wärme-Ausdehnung hat.

Ich würde diese massiven Sicherheitsaufschläge rausnehmen. Wer mehr Abstand will, kann ja einfach die Schraube weiter rausdrehen. Conrad empfiehlt für den RF1000 einen Abstand von 0.3 mm. Was der Code aktuell will ist was in der Gegend von 1mm, also 3 mal so viel! IMO sollte sich der Scan da an die Emfehlung von Conrad halten. Meinetwegen kann er sich den hïchsten Punkt in der Matrix raussuchen und den Scan dort durchführen. Was ich aber definitiv erwarte, dass auf dem Display steht, wie groß der Abstand ist, und nicht irgendwelche magischen Zahlen, die keiner versteht ohne den Quellcode zu lesen.

Also konkret, spricht was dagegen, wenn ich das so ändere?

Nibbels, 2020-06-27:
Richtig cool wäre noch, wenn man  die Ausdehnungskonstanten im EEPROM oder Druckermenü einstellen könnte. ^^
Aber das ist glaube ich etwas Overkill. Fürs V6 bzw. Fullmetal müsste man die mal messen. Die haben ja kein PEEK drin.

mhier, 2020-06-29:
Also erstmal: Die 0.3mm von Conrad sind für einen *kalten* Scan gedacht. Das hat eigentlich immer gut funktioniert. Klar haben manche Leute sich die Düse mal auf dem Heizbett abgeschliffen, aber das ist dann eben eine Fehlbedienung. Die Fehlbedienung mag unwahrscheinlicher werden, wenn man die Schraube sehr weit weg einstellt, aber man kann auch ganz einfach die Fehlbedienung selbst vermeiden. Dafür braucht es IMO keine riesigen Aufschläge. So ganz ist mir auch gar nicht klar, wogegen das eigentlich schützen soll.

Ich führe den Z-Offset-Scan grundsätzlich kalt durch, das funktioniert besser bei mir. In dem Fall rechnet der Scan mit (EXTRUDER_MAX_TEMP-20)*0.95 = 242um Längenausdehnung. Hinzu kommt die "Nachdehnung" (was auch immer das sein soll - physikalisch eine sehr fragwürdige Größe) von 150um pauschal. Sind wir schon bei 392um. Dann kommt noch die Ausdehnung des Bettes oben drauf (verrückterweise mit einem größeren Ausdehnungskoeffizienten als das Hotend, das kann physikalisch mal wieder nicht sein) von (120-20)*1.5um = 150um. Damit sind wir bei 542um für "MinDistanceInMikronsWarm". Da drauf wird dann noch RestAbstandInMikrons = 50um addiert, also knapp 600um. Das ganze wird auf das Matix-Maximum addiert und das ist dann mein Zielwert "SollkorrekturWarm". Wenn das Matix-Maximum > 200um ist, hat man verloren, weil der Zielwert dann > 0.8mm ist und damit nicht erreichbar.

SollkorrekturWarm sollte unter diesen Bedingungen (kalter Extruder, kaltes Bett) laut Anleitung von Conrad 0.3mm sein. Wir sind hier bei einem perfekt ebenen Bett bei dem doppelten, in Realität also sogar noch mehr.

Oder versteh ich da was falsch? Es dekt sich jedenfalls mit meiner Beobachtung. Der Scan sagt mir, ich soll die Schraube so weit rausdrehen, dass ich kalt das Bett gar nicht mehr berühren kann. Damit kann ich keinen HBS mehr durchführen. Das ist unbrauchbar!

An den Koeffizienten schrauben hilft nur begrenzt. Schließlich soll das ja auch mit dem V2 funktionieren. Ich glaube vielmehr, die Grundidee ist falsch. Korrigier mich, wenn ich mich irre, aber es geht doch nur darum, zu verhindern, dass jemand versehentlich mit der Düse über das Bett schleift? Das sollte man doch wenn dann an ganz anderer Stelle verhindern. Z.B. könnte man einen Schutzmechanismus einbauen, der Bewegungen in X/Y nicht zulässt, wenn der korrigierte (also abgeschätzt wahre) Abstand zwischen Düse und Bett zu gering ist (auch dann, wenn die Korrektur aus ist natürlich). Und Homing in Z wird nicht ausgeführt, wenn der Tisch nicht bei Y<10mm oder so steht. Das wären brauchbare Sicherheitsmaßnahmen, die die Funktion nicht beeinträchtigen würden. Beides wird durch eine weit rausgedrehte Schraube gar nicht zuverlässig verhindert (jedenfalls solange ich mit der Konfiguration noch drucken kann), z.B. kann man im Slicer nen Z-Offset von -1mm einstellen und schon geht's schief. Das Einzige, was vielleicht verhindert wird, ist, dass man sich die Düse kaputt macht, wenn man die Z-Kompensation nicht an hat und dann versucht zu drucken. Aber in dem Zustand kommt eh kein brauchbares Ergebnis zustande. Man wird i.d.R. direkt feststellen, dass da was nicht stimmt, bevor die Düse kaputt geht. Und wer bei neuen Druckeinstellungen nicht wenigstens am Anfang zuschaut, dem ist halt nicht mehr zu helfen...

Also nochmal die Frage: was genau soll das Ganze bezwecken? Ich will ja nicht irgendwas rausscheißen, was doch irgend einen Sinn hat, aber ich seh den einfach nicht...

mhier, 2020-06-29:
Achja: Den Z-Offset-Scan warm durchführen ist vielleicht eine Sache, den HBS-Scan müsste man dann aber ebenfalls warm durchführen. Die allererste Idee vom Z-Offset-Scan war es doch gerade, genau das zu vermeiden!

mhier, 2020-06-29:
Ok andere Idee:

Wir lassen die Berechnung so drin wie sie ist. Es gibt aber nach der Berechnung noch eine Prüfung, dass der Zielwert keinesfalls dazu führen darf, dass der tiefste Punkt des Heizbettes nicht mehr durch die Düse im kalten Zustand erreichbar ist. Der Abstand tiefster Punkt vom Heizbett zum Z-Schalter darf also kalt nicht größer sein als Z_ENDSTOP_DRIVE_OVER minus sagen wir 0.1mm (damit man da auch zuverlässig beim Z-Offset-Scan hinkommt, auch wenn die aktuelle Matrix nicht ganz richtig ist).

Dann können Besitzer eines RF2000 mit einem stärker überfahrbaren Z-Schalter die Sicherheit gegen Fehlbedienung behalten, wer einen Original-RF1000 hat, wird aber nicht in die Wüste geschickt :-) Dann wird die Schraube eben so weit wie möglich rausgefahren, aber auch nicht weiter.

Passt das?

Nibbels, 2020-07-01:
Ich muss deine Kommentare die nächsten Tage nochmal in Ruhe lesen. Gerade ist mein Kopf zu voll.
Der letzte Punkt scheint für mich aber absolut schlüssig. Erinnere ich mich richtig, beachten wir immer den obersten Matrixpunkt(x,y)_max.

Wir erhalten also ein Ergebnis. Dann müssen wir den untersten Matrixpunkt(x,y)_min prüfen, ob er unter Z_ENDSTOP_DRIVE_OVER liegt und das was "übersteht" als Ergebnis nach oben korrigieren. Sodass der unterste Matrixpunkt wieder innerhalb der Varfahrtoleranz des Schalters liegt.
Sollte man die Korrektur machen könnte man sich noch überlegen, eine veränderte Nachricht auszugeben.

Soweit klar.

Ich glaube übrigens dass wir historisch erst den Schraubenscan und dann die exakten 0.8mm festgelegt haben. Und 0.8mm auch nur Pi*Daumen: Es kann durchaus sein, dass wir auch 1mm reinschreiben können und alles ok ist. Aber das weißt du u.U. besser.
Ich könnte höchstens mit meinem Max-Endstop mal testen. Oder findet man Datenblätter zu den kleinen Buttons?

mhier, 2020-07-02:
Gab's das Z_ENDSTOP_DRIVE_OVER nicht schon in der original Conrad-Firmware?

closed 2023-03-22 by mhier